ros-industrial/universal_robot #155: Question about IKFast & UR arms

TheDash, 2015-01-23:
Does there exist a working implementation of OpenRave's IKFast system for UR10 & UR5 arms? Just curious before I start hacking away on one!

shaun-edwards, 2015-01-23:
We don't have one for a bare arm (all of ours have an end-effector on them).  I can say that ikfast is an order of magnitude faster than KDL.

TheDash, 2015-01-23:
I see. And yes, KDL is causing some difficulties within MoveIt. Do you know if there's an available IKFast for MoveIt? I saw the moveit_ikfast package, but is there any kind of configuration I'll need to do for the UR10?

shaun-edwards, 2015-01-23:
It's probably easiest to re-run the moveit setup assistant on the existing package.  The drop down for kinematic chains should show any/all IK plugins.

Otherwise, you could edit the `kinematics.yaml` file to point to the fast IK plugin package.  The fanuc packages make use of ikfast (see [here](https://github.com/ros-industrial/fanuc/blob/hydro-devel/fanuc_lrmate200ic5l_moveit_config/config/kinematics.yaml)).

TheDash, 2015-01-23:
Ah ok, so I do have to configure/create one for the UR10 arm then.

gavanderhoorn, 2015-01-23:
@TheDash: does it have to be ikfast based? The repository already contains IK/FK packages for both the UR5 and the UR10. In principle I'd say they should offer the same (or even better, see [here](http://openrave-users-list.185357.n3.nabble.com/OpenRAVE-IKFast-Failure-UR10-td4026158.html)) level of functionality, if we can get the issues that some users have reported cleared up (#127).

It would perhaps be more efficient to try and fix that, than starting from scratch with IKFast (which can be difficult for the URs).

ipa-fxm, 2015-01-23:
In the [kinematics.yaml](https://github.com/ros-industrial/universal_robot/blob/hydro-devel/ur5_moveit_config/config/kinematics.yaml) there is already a commented block that would use the analytical IK solver from the [ur_kinematics](https://github.com/ros-industrial/universal_robot/tree/hydro-devel/ur_kinematics) package (it's also there for the UR10).

You can try it, but as @gavanderhoorn mentioned this solver will return IK solutions that are within the full joint range [-2Pi...2Pi] so that it only works with the ur_description with the limited option set to false, with which MoveIt! cannot plan in a reasonalbe amount of planning time....

There have been tests with IKFast and the universal arms a while ago (groovy times), but the solver did not return correct solutions in some cases. This was actually the reason why work on an analytical IK solver has been started by @kphawkins.

Btw, you can find the IKFast related works [here](https://github.com/ipa320/cob_manipulation/tree/indigo_dev/cob_kinematics). I don't think they will compile under indigo, but if you want to have a look at it....

de-vri-es, 2015-01-23:
If that's the only problem, wouldn't it be easy to wrap the analytical solver in ur_kinematics to return only solutions in [-Pi, Pi] or [0, 2Pi] (or any 2Pi range)?

gavanderhoorn, 2015-01-23:
@de-vri-es: I'm not really sure that the issue really is with the `ur_kinematics` solvers.

The plugins do have the *load-joint-limits-from-urdf* bit of obligatory code in them ([ur_kinematics/src/ur_moveit_plugin.cpp - lines 217 to 246](https://github.com/ros-industrial/universal_robot/blob/hydro-devel/ur_kinematics/src/ur_moveit_plugin.cpp#L217-L246)), as all IKFast plugins do. I figured @kphawkins would be much faster in fixing this, so I never looked any further.

gavanderhoorn, 2015-01-23:
@shaun-edwards wrote:
> We don't have one for a bare arm (all of ours have an end-effector on them) [..]

I've always thought it made sense to create IKFast plugins for just the kinematic chain from base to end-of-arm, excluding any tooling. That would then make it possible to re-use them in any situation. Including the EOAT in your IKFast plugin seems like a strange thing to do?

shaun-edwards, 2015-01-23:
@gavanderhoorn - strange indeed ;)

TheDash, 2015-01-23:
Conclusion from this discussion: IK/FK exists for UR arms. But are broken and slow in unlimited joint space and does not work in limited joint space. 

Actions: fix the IK/FK so that it's better in unlimited joint space? is that an inherent problem of the algorithm being used or the programming is wrong for it?

Should we be porting this to IKFast if that works, or looking for different optimization algorithms?

gavanderhoorn, 2015-01-23:
> Conclusion from this discussion: IK/FK exists for UR arms. But are broken and slow in unlimited joint space and does not work in limited joint space. 

Not at all.

The algorithm implemented in `ur_kinematics` are far from slow. They might be a little slower than IKFast generated plugins, but not much, if at all. The only issue with them is that they *seem* to ignore joint limits, and only when a `limited` version of the URDFs is loaded.

imho, the best way to deal with this is contact @kphawkins, and figure out what is going on with `ur_kinematics`.

TheDash, 2015-01-23:
My apologies, 

"full joint range [-2Pi...2Pi] so that it only works with the ur_description with the limited option set to false, with which MoveIt! cannot plan in a reasonalbe amount of planning time...."

"And re: "are slow": as far as we know, it's MoveIt that has an issue with joint limits that are > PI. That is what @ipa-fxm was referring to."

Thanks, that clarified it. URKinematics plugin + limited solver. I'll give it a go and contact kphawkins. This thread has been super helpful and informational on the matter. Is this information made available for new users? E.g suggestions of what configuration to use and to comment out that plugin for improved performance?

kphawkins, 2015-01-23:
My kinematics solutions are far faster than even the IKFast solutions. I
think it only takes around 7us to get the full set of solutions.
Furthermore, the IKFast implementation was broken last time I checked. They
might have fixed it, but it's the sort of problem which is both hard to fix
and detect. Certain configurations, like having the gripper upright, fail
to give a solution.

My Moveit plugin does work for some functionality, but seems to fail with
planning or something due to some assumptions in MoveIt. I haven't had the
time to investigate, but it sounds like Felix might have even more
experience than me.

I'd be okay inserting code which does the type of angle wrapping Moveit
expects into the plugin code. You'd probably put it right here:
https://github.com/ros-industrial/universal_robot/blob/hydro-devel/ur_kinematics/src/ur_moveit_plugin.cpp#L648
I'm not sure exactly what needs to be done, but it sounds like we just
modify the range of the solutions. You might need to also alter the seed
value in the previous line which takes a value for the last joint angle.
This is used in the rare singularity where joints 2, 3, and 4 are parallel
with joint 6 and we get infinite solutions.

Also, to comment on some extended functionality of that IK, it should work
even if the arm is in the middle of a longer kinematic chain. That is, if
it has joints either before the base or after the gripper. At the moment,
the way I solve the IK is to randomly sample joint configurations of the
other joints until I can find a solution for the arm, then return that.
However, this could easily be turned into an optimization routine where it
searches over the extra joints for the "best" IK solution. This should also
be fairly easy to implement and would go at the same place in the code as
above. You might need to be careful with the loop logic though.

My apologies,

"full joint range [-2Pi...2Pi] so that it only works with the
ur_description with the limited option set to false, with which MoveIt!
cannot plan in a reasonalbe amount of planning time...."

"And re: "are slow": as far as we know, it's MoveIt that has an issue with
joint limits that are > PI. That is what@ipa-fxm
<https://github.com/ipa-fxm>was referring to."

Thanks, that clarified it. URKinematics plugin + limited solver. I'll give
it a go and contact kphawkins. This thread has been super helpful and
informational on the matter. Is this information made available for new
users? E.g suggestions of what configuration to use and to comment out that
plugin for improved performance?

ipa-fxm, 2015-01-28:
I investigated this a bit...(without diving too much into the actual IK solver code)!
In my [branch](https://github.com/ipa-fxm/universal_robot/tree/investigate_ur_kinematics), I added some more output to see what's going on in the ```searchPositionIK``` function.

What I can see is that the joint_limits are correctly represented in the robot_model and the [robot_state](https://github.com/ipa-fxm/universal_robot/blob/investigate_ur_kinematics/ur_kinematics/src/ur_moveit_plugin.cpp#L318). So this is correct!

But what I see is that the [ik_seed_state](https://github.com/ipa-fxm/universal_robot/blob/investigate_ur_kinematics/ur_kinematics/src/ur_moveit_plugin.cpp#L560) contains values that are out of limits!

You can reproduce the problem (and my tests) by simply running
```
roslaunch ur5_moveit_config demo.launch limited:=true
```
from the branch mentioned above.
If you then move the interactive marker to specify a planning goal, you will see the orange model for the goal state turning pink, which means that joint_limits are violated...

@kphawkins: 
Do you have an idea where the ```ik_seed_state``` comes from and how to guarantee that it lies within the joint_limits specified in the ```joint_model_group_``` (limited or unlimited)?

kphawkins, 2015-01-28:
That depends on who's calling it. If those values are out of joint range,
then it is the higher level code which is to blame. You always need a seed
state to disambiguate which IK solution you want. I think my solver tries
to find a solution which satisfies joint limits, but does so by using the
built in solution checker.

I'm seeing a problem in this code though. It doesn't attempt to check the
alternate joint configurations. That is, on top of there being the 8 joint
configurations possible for most IK solutions, each joint can be flipped
2pi in one direction for the other solution (giving a total of 512
solutions). You have to resolve which one you want using the seed state or
something else. Checking all solutions is probably really time consuming.

But if the seed state is wrong, I don't know if fixing that will help. That
previous fix of altering the joint range might be a much more palatable
solution at the moment.

TheDash, 2015-02-17:
Thanks @kphawkins . I tried this configuration out today and it's really fast.

ipa-fxm, 2015-02-18:
@TheDash 
Is this issue really fixed? I mean do you have a ur_kinematics IKFast solver that is able to handle the limited versions of the UR?
If yes: please send a PR!
If no: please reopen!

TheDash, 2015-02-18:
@ipa-fxm,

I set the kinematics solver to be the UR5KinematicsPlugin and the joint_limited:=true

It works fast and optimally (except for the joint limit issue sometimes causing to appear not to be optimal)

ipa-fxm, 2015-02-18:
So you can use the UR5KinematicsPlugin although it returns IK-solutions that are beyond the __joint_limited__ ranges of the URDF?
Are you using MoveIt! from the rviz-plugin or using the moveit_commander API?

TheDash, 2015-02-18:
I'm not sure if the solutions are beyond the joint_limited ranges. I'm using the MoveIt! Rviz plugin and selecting "Random update" which picks a working configuration inside the possible joint limits and then "Plan and execute" which executes and plans in under .5s

I do note that since the joints are limited, I can't select a free-form configuration (E.g drag the tip of the arm to a place) because it goes pink, showing that its joints are out of range.

When using the joint_limited=false, the URXKinematics solver never finds a solution.

gavanderhoorn, 2015-02-18:
On 18-2-2015 11:08, Devon Ash wrote:
> I'm not sure if the solutions are beyond the joint_limited ranges. I'm using the MoveIt! Rviz plugin and selecting "Random update" which picks a working configuration inside the possible joint [...]

Note that, as far as I know, using 'random valid' in the MoveIt planning 
plugin doesn't use IK. It just does a random sampling for all the 
joints, then plans a joint motion between where it is right now (current 
state) and the target (in joint space).

ipa-fxm, 2015-02-22:
I think I have a solution that allows to use the ur_kinematics moveit_plugins with the limited version of the URs:
You can find it in my [branch](https://github.com/ipa-fxm/universal_robot/compare/ros-industrial:indigo-devel...investigate_ur_kinematics).
It's still WIP and still has quite some debug output that will be removed again...that's why it's not a PR yet!
But I'd like to hear your opinions about it!

In the end, this is the main [commit](https://github.com/ipa-fxm/universal_robot/commit/7ae155809f359015f4118fe4a5d06d8b75d0c731)
I added a compiler flag triggered angle wrapping so that I end up with four ur_kinematics moveit_plguins: UR5, UR5Limited, UR10, UR10Limited
The respective usage of the various plugins within the kinematics.yaml of the moveit_configs can be seen in this [commit](https://github.com/ipa-fxm/universal_robot/commit/ec369a798cba22be2e434447e45b8b34225ea839)

You could also use the Limited moveit_plugins with the non-limited versions of the arms....
It seems to work with the InteractiveMarker in the rviz plugin.

However, would this be the solution we want?
@kphawkins Is there any other chance to modify [ur_kinematics::inverse()](https://github.com/ipa-fxm/universal_robot/blob/7ae155809f359015f4118fe4a5d06d8b75d0c731/ur_kinematics/src/ur_kin.cpp#L163) to work with the limited version?

Please test and comment here!

ipa-fxm, 2015-03-05:
Any comments on my test branch? Should I create a PR for this? I'll remove the debug output of course...

gavanderhoorn, 2015-03-05:
I'm not too much of a fan of the `#ifdef`s, plus the additional kinematics plugins. I get why you chose to do it this way though.

I also don't really understand why this is even necessary, if the joint limits are loaded from the parameter server at plugin initialisation?

kphawkins, 2015-03-05:
I think this might be a decent solution if you're happy with it. Looking
back on this, the IK should consult the joint limits read in from the URDF
to insure the joint flipping is done correctly. You can find them being
read in here:
https://github.com/ros-industrial/universal_robot/compare/indigo-devel...ipa-fxm:investigate_ur_kinematics#diff-6f25384c6c7d94a33c6492d5e4083c09L224

Using that should remove the need for compiling separate implementations.
The logic is simple, just add and subtract 2pi to find which (0-2) of the
three are in the limits, and find the closest to the seed state. I think
that should guarantee that you select the best joint value for that joint
configuration (where this configuration is one of the 8 you typically get
from joints limited to [-pi,pi] ).

This solution might be complete if the behavior of the collision detection
method is unchanged by rotations like this. Else you might need to modify
the loop to check configurations which are further from the seed state with
joints rotated by 2pi are satisfactory.

Any comments on my test branch? Should I create a PR for this? I'll remove
the debug output of course...

ipa-fxm, 2015-03-06:
Thanks for the comments...see #184

TheDash, 2015-03-10:
So of all this discussion; what's the update on having a planner as good as the IKFast but with unlimited joints?

ipa-fxm, 2015-03-10:
The pending PR #184 fixes the analytical IK solver provided in ur_kinematics to correctly consider the joint limits as specified in the URDF. Thus, it returns correct results for both limited and unlimited versions of UR5/UR10!

TheDash, 2015-03-10:
I'll give this PR a go on our real UR5 arm and let you know how performance is.

TheDash, 2015-03-11:
Sorry about asking, but I'm just trying to understand the issue here. Before, I tried using the unlimited setup with the URKinematics planner and nothing ever returned from plan (but I could freely move the arm in MoveIt)

I can use the URKinematics planner in MoveIt with a limited joint configuration now, and it works well, but the orb in moveit doesn't allow me to freely move the arm around (it must be within joint ranges and makes the arm purple). 

Does this proposed change allow me to freely move the orb around with a limited joint setup?

ipa-fxm, 2015-03-11:
I don't understand your problem.
With #184 you can use the URKinematics plugin with both limited and unlimited versions!
Dragging around the InteractiveMarker for the GoalState should never turn purple anymore!

gavanderhoorn, 2015-03-12:
@TheDash wrote:
> Does this proposed change allow me to freely move the orb around with a limited joint setup?

It should. If it doesn't, please report back as that would be a problem.

abubeck, 2015-03-12:
As #184 was merged I will close this issue. @TheDash: if you have further problems open a new separate issue

closed 2015-03-12 by abubeck