ros-industrial/universal_robot #158: Improving the test_move script

Jmeyer1292, 2015-01-29:
I recently wrote a tutorial on getting started with the Universal Robots and ROS-I here: http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial. Most of this was copy/paste from the readme pages here on Github.

I realize this is probably low priority, but I believe that the current test_move script executes a movement that is too aggressive. I know I was surprised by it and I know others with similar feelings. As this file is the first introduction many will have to the programmatic control interface provided by the ur_driver, I believe that it is important that we improve this file.

Specifically, I suggest a small menu with a choice of different movements and an explanation of what they are. Or perhaps simply a smaller test movement that mostly manipulates the last two joints.

gavanderhoorn, 2015-01-30:
The test script is a relic from the earliest days of the driver (see a0b1c2c9). I don't think it was ever changed since then. I fully support changing its behaviour to be more predictable and / or updating documentation. Perhaps we could even include a short video, so users know what they can expect.

gavanderhoorn, 2016-12-30:
@Jmeyer1292: have you had any time to work on this?

ipa-hsd, 2018-07-11:
https://github.com/ros-industrial/universal_robot/pull/366