ros-industrial/universal_robot #159: Action client not connected

psfafz, 2015-02-06:
Hi,

I have the indigo version of the ur5 package using a 01/2010 roboter version with the URControl version 1.5.7849.

I can properly connect to the robot typing:
roslaunch ur_bringup ur5_bringup.launch limited:=true robot_ip:=IP_OF_THE_ROBOT [reverse_port:=REVERSE_PORT]

however, when I try to run the moveit planner typing:
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

I get the error messange:

[ERROR] [....]: MoveitSimpleControllerManager: Action client not connected: /follow_joint_trajectory

How can I fix this issue?
This issue, by the way, does not appear on the indigo-devel version. I am not using this version since I have some other issues there

thanks for helping

ipa-fxm, 2015-02-09:
@psfafz 
Hmm...I don't have a UR5 v1.5 at hand...but I've got the following comments:
- Support for the parsing of the RealtimeProtocol for v1.5 robots has only been added [recently](https://github.com/ros-industrial/universal_robot/commit/748c5227d44fdc6e3626dcfbdc764e7d3a16512a) and is only available in indigo-devel, yet
- However, this should only affect the availability of the /joint_states topic...none of the differences between indigo and indigo-devel are related to the name of the follow_joint_trajectory action or the moveit_configs...thus, the action ```/follow_joint_trajectory``` should be available...but maybe the MoveitSimpleControllerManager requires the joint_state topic to be available somehow

In your setup can you try to find the follow_joint_trajectory action with ```rostopic  list```?
What kind of problems are you facing with the indigo-devel branch?

gavanderhoorn, 2015-02-09:
The Action server is instantiated by the driver, which didn't support version 1.5 of Polyscope until recently, as @ipa-fxm mentioned. If you are not using a `indigo-devel` clone, I think the driver will crash, as it cannot deserialise the packet data coming in over the socket connection. If I remember correctly, the driver crashing will also remove the Action server topics, which would result in the error reported.

@psfafz: is the driver still running at the point where you receive the error message? It is already pretty verbose by itself, but you might try `roslaunch --screen ur_bringup ur5_bringup.launch ..` to see if you get any more output.

tnl-woi, 2015-02-10:
Yes, I agree with @gavanderhoorn. If the parser/deserializer fails, the driver will crash and thus will take the action server down. This explains the issue, IMHO.

psfafz, 2015-02-11:
@ipa-fxm
`follow_joint_trajectory` does not appear on the topic list. Before prompting the error as described above I get `[INFO] [...]: MoveitSimpleControllerManager: Waiting for /follow_joint_trajectory to come up` twice.

For the sake of readability I will open a new issue for my indigo-devel problem since it is not related to this topic

@gavanderhoorn 
This is the output I get when launching the controler:

...
[ INFO] [1423650023.563496810]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1423650023.563525769]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1423650023.563552833]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1423650028.665023275]: MoveitSimpleControllerManager: Waiting for /follow_joint_trajectory to come up
[ INFO] [1423650033.665262262]: MoveitSimpleControllerManager: Waiting for /follow_joint_trajectory to come up
[ERROR] [1423650038.665622513]: MoveitSimpleControllerManager: Action client not connected: /follow_joint_trajectory
[ INFO] [1423650038.687841062]: Returned 0 controllers in list
[ INFO] [1423650038.698525098]: Trajectory execution is managing controllers
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
[ INFO] [1423650038.905875967]: 

********************************************************
* MoveGroup using: 
*     - CartesianPathService
*     - ExecutePathService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
*     - GetPlanningSceneService
********************************************************

[ INFO] [1423650038.905946769]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1423650038.905963303]: MoveGroup context initialization complete

All is well! Everyone is happy! You can start planning now!

gavanderhoorn, 2015-02-11:
@psfafz: the output you copy seems to be from one of the launch files in the MoveIt configuration pkgs.

I'm more interested in the output of the `roslaunch ur_bringup ..` command. On my machine, I get something like:
```
[ skipped standard roslaunch output]

setting /run_id to 9252166e-b1db-11e4-bf64-d8d385994de6
process[rosout-1]: started with pid [1277]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [1289]
process[ur_driver-3]: started with pid [1307]
process[tf2_buffer_server-4]: started with pid [1308]
Setting prefix to 
[WARN] [WallTime: 1423651735.463263] No calibration offset for joint "shoulder_pan_joint"
[WARN] [WallTime: 1423651735.463572] No calibration offset for joint "shoulder_lift_joint"
[WARN] [WallTime: 1423651735.463777] No calibration offset for joint "elbow_joint"
[WARN] [WallTime: 1423651735.463982] No calibration offset for joint "wrist_1_joint"
[WARN] [WallTime: 1423651735.464216] No calibration offset for joint "wrist_2_joint"
[WARN] [WallTime: 1423651735.464583] No calibration offset for joint "wrist_3_joint"
[INFO] [WallTime: 1423651735.464992] No calibration offsets loaded from urdf
[INFO] [WallTime: 1423651735.466226] Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [WallTime: 1423651735.470758] Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [WallTime: 1423651735.519617] Programming the robot
[INFO] [WallTime: 1423651735.520657] Programming the robot at 127.0.0.1
Handling a request
[2015-02-11 11:48:55.802216] Out: hello
[INFO] [WallTime: 1423651735.834655] Robot connected
The action server for this driver has been started
```

Just to be clear: are you starting things following the steps in the [readme](https://github.com/ros-industrial/universal_robot/blob/hydro-devel/README.md)?

psfafz, 2015-02-11:
@gavanderhoorn 
I do follow the steps in the readme.
This is the output I get from the ur_bringup launch file:

[WARN] [WallTime: 1423652269.050839] No calibration offset for joint "shoulder_pan_joint"
[WARN] [WallTime: 1423652269.051036] No calibration offset for joint "shoulder_lift_joint"
[WARN] [WallTime: 1423652269.051170] No calibration offset for joint "elbow_joint"
[WARN] [WallTime: 1423652269.051303] No calibration offset for joint "wrist_1_joint"
[WARN] [WallTime: 1423652269.051429] No calibration offset for joint "wrist_2_joint"
[WARN] [WallTime: 1423652269.051564] No calibration offset for joint "wrist_3_joint"
[INFO] [WallTime: 1423652269.051733] No calibration offsets loaded from urdf
[INFO] [WallTime: 1423652269.052538] Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [WallTime: 1423652269.053771] Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [WallTime: 1423652269.074480] Programming the robot
[INFO] [WallTime: 1423652269.075242] Programming the robot at 192.168.1.1
RobotStateRT has wrong length: 756
Exception in thread URConnectionRT:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/faps/catkin_ws/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 374, in __run
    self.__on_packet(packet)
  File "/home/faps/catkin_ws/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 340, in __on_packet
    for i, q in enumerate(stateRT.q_actual):
AttributeError: 'RobotStateRT' object has no attribute 'q_actual'

Handling a request
[2015-02-11 11:57:49.868428] Out: hello
[INFO] [WallTime: 1423652269.876088] Robot connected

gavanderhoorn, 2015-02-11:
Ok. Then it is exactly as I described in my earlier comment: the driver is crashing, as the version you are using cannot deserialise the V1.5 binary protocol. This also downs the action server.

Solution: either wait for `indigo-devel` to be released, or run the entire `universal_robot` stack from source in a catkin workspace.

gavanderhoorn, 2015-02-13:
@psfafz: are things working for you with `indigo-devel`? Your comments in #160 seems to suggest they do.

If so, please close this issue.

closed 2015-02-13 by psfafz

ChunJyeBehBeh, 2019-03-07:
Hello,

I facing the same probelm when I run the code on Ubuntu 16.04 ros Kinetic.

```
[ WARN] [1551942153.489598473]: Waiting for /vel_based_pos_traj_controller/follow_joint_trajectory to come up
[ WARN] [1551942159.490069471]: Waiting for /vel_based_pos_traj_controller/follow_joint_trajectory to come up
[ERROR] [1551942165.490394613]: Action client not connected: /vel_based_pos_traj_controller/follow_joint_trajectory
```

May I know how you solve it?