ros-industrial/universal_robot #160: Crash after connection with warehouse server

psfafz, 2015-02-11:
I have the indigo-devel version of the ur5 package using a 01/2010 roboter version with the URControl version 1.5.7849.

I can properly run the driver, the moveit planner and the rviz interface - everything comes up without any errors. I can plan and excecute on the real robot.
First thing to notice is that the top bar of the rviz window flickrs right from the start (meaning the bar "moveit.rviz -RViz").
 I can connect to the warehouse server after typing 
`roslaunch ur5_moveit_config warehouse.launch moveit_warehouse_database_path:=~/moveit_db`
After loading a stored scene, the scene objects flickers in orange/yellow, and the fps rate drops significatnly so that after a couple of seconds my machine freezes entierly.

gavanderhoorn, 2015-02-11:
Could you update your OP with info on your OS and hardware? I'm not saying it isn't possible, but I'd be very much surprised if this is actually an issue with the `universal_robot` packages.

Flicking / rendering issues typically point towards problems with graphic cards & 3d acceleration.

gavanderhoorn, 2015-02-11:
> First thing to notice is that the top bar of the rviz window flickrs right from the start 

If you start rviz in isolation, do you get flickering as well?

What happens when you start the following:
```
roslaunch ur_description ur5_upload.launch
roslaunch ur_description test.launch
```

psfafz, 2015-02-11:
@gavanderhoorn 
On the same machine, rviz runs stable and without flickering with the indigo `universal_robot` package
No problems when running rviz in isolation either.

There are no problems with flickering or instabilities either when starting:
```
roslaunch ur_description ur5_upload.launch
roslaunch ur_description test.launch
```

gavanderhoorn, 2015-02-11:
Just to see whether this is something specific to the ur pkgs: could you clone the [fanuc](https://github.com/ros-industrial/fanuc) repository to your workspace, build it and then run `roslaunch fanuc_lrmate200ic_moveit_config demo.launch db:=true` (apart from the driver, this starts all the same nodes as the ur launch files you are using)? You could also provide `db_path`, and set it to your current mongodb path.

If I read your OP correctly, the flickering should already start just after loading moveit & rviz right?

Please report whether you see the same issues there.

psfafz, 2015-02-13:
I don't see an indigo version of the fanuc repository, and I can not build the hydro version on my ros-indigo.

> If I read your OP correctly, the flickering should already start just after loading moveit & rviz right?

yes you are right

gavanderhoorn, 2015-02-13:
> I don't see an indigo version of the fanuc repository, and I can not build the hydro version on my ros-indigo.

There is no Indigo version, but the `hydro-devel` branch should build on Indigo installations. If it is really an issue with the Fanuc packages, please report it on the [issue tracker](https://github.com/ros-industrial/fanuc/issues).

gavanderhoorn, 2015-02-13:
Btw: it's not really important which particular MoveIt configuration you use, I just thought the ones in the Fanuc stack would be easiest to get going (I didn't realise they require building both `fanuc` and `industrial_core` from source). If you have any other MoveIt configurations, you can use those as well. All we need to do is establish whether this is really an issue with the `universal_robot` packages, or more general MoveIt or graphics hardware related.

psfafz, 2015-02-13:
On the same machine,  rviz was running with the `univeral_robot` indigo version without flickering. In addtion, this machine was used to control a baxter which was running without any problems.

gavanderhoorn, 2015-02-13:
All changes between v1.1.5 and current `indigo-devel` are listed [here](https://github.com/ros-industrial/universal_robot/compare/1.1.5...indigo-devel).

I don't see anything related to the visual parts, let alone the MoveIt packages.

Have there recently been any updates to the kernel, drivers or other software of the machine in question?

ipa-fxm, 2015-02-14:
I absolutely cannot think of this problem being universal_specific!
If you are having problems with using the moveit_config in the fanuc repository, simply try to create your own moveit_config package for any URDF you have access to using the MoveIt! SetupAssistant then try the warehouse.launch of that config package.
Maybe it has something to do with the stored scene you are trying to load from your warehouse database...

gavanderhoorn, 2015-02-16:
@psfafz: could you contact me via email please?

psfafz, 2015-02-16:
I solved this issue just this morning. I used a computer with dedicated graphics card. This helped me solving all my issues. No more flickering or crashes. Thanks a lot for helping and for continuing this project. I appreciate it very much and I do like this new indigo-devel a lot. If you need any more information about my crashes just contact me, otherwise I will close this issue soon

gavanderhoorn, 2015-02-16:
That's great, but I'd still appreciate it if you could contact me via email (I couldn't find your email address anywhere).

gavanderhoorn, 2015-02-16:
re: closing this issue. If your issue was resolved, then please close it. We can mark it `worksforme`, as it seems it was a user configuration issue.

closed 2015-02-16 by psfafz