ros-industrial/universal_robot #162: 'prevent_programming' should be set using dynamic_reconfigure, not by polling the parameter server

gavanderhoorn, 2015-02-13:
Even though it's only [accessed](https://github.com/ros-industrial/universal_robot/blob/1d99ea831dae0259634d9389c1ee42690e9c3d82/ur_driver/src/ur_driver/driver.py#L968-L974) when the Python side loses connection to the controller (and on initial upload), polling the parameter server for the value of `prevent_programming` doesn't seem very nice. 

The proper alternative would seem to be adding a dynamic_reconfigure server to the driver.

ipa-fxm, 2015-02-14:
I started this on [my branch](https://github.com/ipa-fxm/universal_robot/tree/prevent_programming).
I need to test it with the hardware - thus, no PR yet...

ipa-fxm, 2015-02-16:
see #163

ipa-fxm, 2015-02-20:
see #170

abubeck, 2015-03-12:
Dynamic reconfigure is in (#163). Therefore I will close this.
Discussion on having both (rosparam and dynamic_reconfigure) is in #170

closed 2015-03-12 by abubeck