ros-industrial/universal_robot #171: Modifying meshes

scottp11, 2015-02-22:
Hi, 

I want to modify the visual mesh for the base of the UR5. What program can I use to do this? I have tried  blender, meshlab, and sketchup. 

Kind regards, 

Scott

de-vri-es, 2015-02-22:
Hey Scot,

Recent versions of Blender can work with collada (.dae) files. I'm not sure when the feature was included (might be 2.62), but you may have to use a more recent version than your distribution provides. You also have to select the import option from the file menu.

gavanderhoorn, 2015-02-23:
@scottp11: @de-vri-es is right, Blender is probably your best choice (although there are other programs that can edit Collada meshes).

Afaik, meshlab doesn't support per vertex/face/edge editing of meshes.

Sketchup should also be able to edit Collada files, but depending on your setup, you might run into scale issues (ROS uses meters for everything, which is not always the default for many modelling tools).

From your comment it would seem that you have not been successful with any of the programs you mentioned; perhaps if you can tell us what didn't work, we might be able to help.

PS: in the future, you might want to post these kind of questions on [ROS Answers](http://answers.ros.org/), as it is not a real issue with the package per se.

scottp11, 2015-02-25:
Thank you @de-vri-es and @gavanderhoorn . Blender 2.69 (Ubuntu Software Centre version) of Blender does not support Collada import but 2.73 (from the Blender website) does. 

Steps to modify dae mesh:
1) Install Blender version >= 2.73. As at 26/2/2015 this is done via the Blender website. 
2) Start Blender. Delete the cube at the origin by clicking and hitting delete. 
3) File->import->Collada then select your mesh file. 

There are plenty of resources online describing how to modify your mesh. 

For the record I was removing the black cylinder that protrudes from the UR5 base in the visual model. I don't believe this exists on our robot. I am using the visual model for collision detection as it is in self-collision ~22.3% of the time while the collision model is in self collision >40% of the time. 

Thanks, 

Scott

closed 2015-02-25 by scottp11

gavanderhoorn, 2015-02-26:
That 'black cylinder' is an approximate representation of the signal wire coming out of the robot base.

> I am using the visual model for collision detection as it is in self-collision ~22.3% of the time while the collision model is in self collision >40% of the time. 

That sounds like something is wrong, self collisions should be rare. What are you using to determine those percentages? The MoveIt pkgs have been created such that links that are always adjacent have their collision checks disabled (such as `base_link` and `link_1`).

scottp11, 2015-03-01:
Hi @gavanderhoorn. Sorry my fault.   I have just realised we are using an old version of this package (Aug 2014). Since then the collision meshes have been modified (22 October 2014). We are also using a router end-effector which will increase the number of self collisions. 

I will get back to you if I still see high percentages of self-collisions when we update.

To test self collision percentage I was using robot_state::RobotState::setToRandomPositions and planning_scene::PlanningScene::checkSelfCollision.