ros-industrial/universal_robot #173: ros wiki 'getting started' tutorial incorrectly states that industrial_core is needed

gavanderhoorn, 2015-02-24:
From [Getting Started with a Universal Robot and ROS-Industrial](http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial#Prerequisites) (under Prerequisites):

> 2 . ROS-Industrial's industrial_core package

As far as I know that is incorrect, the `universal_robot` packages don't need this (they don't use the generic clients, nor `simple_message`).

@Jmeyer1292: I think you wrote that tutorial?

Jmeyer1292, 2015-02-24:
I do believe that was a mistake. Still learning. Please delete it.

gavanderhoorn, 2015-02-26:
Done. See [diff/rev2=12/rev1=11](http://wiki.ros.org/action/info/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial?action=diff&rev2=12&rev1=11) for the tutorial page.

closed 2015-02-26 by gavanderhoorn