ros-industrial/universal_robot #176: suggestion: use 'secondary programs' for IO interface

gavanderhoorn, 2015-02-26:
The URScript environment supports the concept of *secondary programs*, that can be send to the controller over port 30002. I propose we use those for the IO setters, instead of the extension that was made to the `prog` URScript program (the additional msg types).

We could (should) still expose the ROS services for users to interact with the driver, but those services should trigger the transmission of a secondary program that performs the requested action.

The `io_states` publisher can still make use of the masterboard data, no changes are needed there.

See the [SecondaryProgram](http://wiki03.lynero.net/Technical/SecondaryProgram) UR wiki article for more info, but basically:
```python
sec secondaryProgram():
  set_digital_out(1, True)
end
```

de-vri-es, 2015-02-26:
Sounds like a good idea.

I have one question though: What happens when you upload multiple secondary programs at the same time?

Ideally both get run fully. If there are any conflicts in setting the same output to different values, then whichever wins doesn't really really matter I'd say.  But suppose other functions are implemented using secondary programs (setting payload weight comes to mind), then it is important that setting outputs and payload weight don't interfere with each other.

I'll run some tests to find out. But in principle +1

gavanderhoorn, 2015-02-26:
> What happens when you upload multiple secondary programs at the same time?

I suggest we just make sure that can't happen. We'll have to track some additional state in the driver, but that shouldn't be too difficult. The only thing I'm not completely sure of right now is implementing some sort of feedback: I've played with `sec` programs before, but never looked at what kind of feedback we can get from them.

If we can't find any reliable way to signal program termination for secondary programs in general, perhaps at least for the IO interface we could try to close the loop by looking at the Masterboard data.

de-vri-es, 2015-02-26:
Well, I really hope it doesn't matter. With a bit of luck there can be multiple secondary programs running at the same time. In practise, there wouldn't be a lot of them running simultaneously anyway, since they just perform a simple task and then quit.

Having ten of them run simultaneously would be an extreme, so if we can verify that there can be 10 of them running without problem, I would say that there is no problem in just sending secondary programs without any form of serializing access.

A quick test with printing messages in a loop on ursim didn't give any meaningful results yet, but I will test more.

gavanderhoorn, 2015-02-26:
It might nice -- from a user perspective -- to be able to tell at least something about the order of execution of programs. Something like 'last setting wins' fi. I'm all for asynchronous-everything, but in my experience that is not true for everyone.

Having some way of reasoning about the execution order of service invocations would be a good thing, imo.

de-vri-es, 2015-02-26:
I see your point, although it wouldn't matter to me personally. When setting conflicting values for the same thing in a short time, it's very likely that at least one of the senders will end up unhappy. If that's the first, the last or a random one is not really important to me.

I did a test where I sent a lot of secondary programs over the same connection to set one output per secondary program. I repeated the test about 20 to 30 times. The effect was that all outputs got set correctly. It's still conceivable that there *might* be race conditions if for some reason the program is executed more slowly, but if so I couldn't trigger any.

So the question then is if serializing the secondary programs is possible and if it is worth the effort. The only use case I can come up with is bit-banging the outputs of the robot arm using service calls. But that seems like a pretty questionable use-case to me.

gavanderhoorn, 2015-02-27:
I agree that it might be a minor point. I just wanted to mention my experience with users that are not so comfortable with systems that aren't synchronous in time.

Good to know that sending multiple secondary programs doesn't pose any obvious problems. Perhaps they are queued internally? Can you say anything about when they started executing?

gavanderhoorn, 2015-02-27:
@de-vri-es wrote:
> I did a test where I sent a lot of secondary programs over the same connection to set one output per secondary program

Ah, over a single connection. Depending on how data is internally buffered, that may be why there were no conflicts. I'd be interested to see what happens with multiple connections (but that may not be a realistic scenario, as a potential implementation would most likely open only a single one).

de-vri-es, 2015-02-27:
> Good to know that sending multiple secondary programs doesn't pose any obvious problems. Perhaps they are queued internally? Can you say anything about when they started executing?

\-----

> Ah, over a single connection. Depending on how data is internally buffered, that may be why there were no conflicts.

What I did was sent all the programs as one blob with netcat. It may even have been a single TCP frame. So using a single connection might already do the trick indeed. It seems to be safe at any rate. So I'm all for this idea.

> I'd be interested to see what happens with multiple connections (but that may not be a realistic scenario, as a potential implementation would most likely open only a single one).

I did try to test opening multiple connections. I ran netcat in the background in a bash script a bunch of times. However, the results where very weird. I expected all netcat instances to run more less simultaneously (since I ran them as background jobs), but that wasn't what happened. I'm not entirely sure what did happen, so that still needs some testing, although it's not directly relevant for the driver.

gavanderhoorn, 2018-06-25:
This is not going to be implemented in `ur_driver`, seeing as it is deprecated and will not be developed any further.

Closing this, but the idea remains valid.

closed 2018-06-25 by gavanderhoorn