ros-industrial/universal_robot #177: suggestion: remove 'io_interfaces.py' from repository/distribution

gavanderhoorn, 2015-02-26:
The `io_interfaces.py` file is a strange file: it uses globals, hard-codes topic and service names (so they are non-remappable), repeats the definition of the `FUN_..` constants, is asymmetric wrt the interfaces it offers (only setters, then uses a callback and variables for access to state), does not guard state variables with locks (so no protection against concurrent access or inconsistent reads) and all setters just wrap ROS service proxies (adding yet another layer of indirection). On top of that: it promotes a style of interaction which is questionable, and we don't have / supply an analogue for the motion interface either.

I get that this is a way of setting up access to Service Proxies that is often used in projects, but I don't think it has its place in the `ur_driver` distribution.

The `test_io.py` script makes use of this file, so if we remove it, `test_io.py` will have to be updated to use its own Service Proxies.

tnl-woi, 2015-02-26:
We already discussed this partly by e-mail.

I certainly agree it's a "strange" file.

I would favor a working API for the I/O interface, in which case ```io_interface.py``` must be rewritten to solve all the issues you mention. However, if the ```ur_driver``` is not the right place for it, then indeed it should be removed and ```test_io.py``` should be updated to at least demonstrate how to use the I/O interface. In this case, not all issues have to be solved (e.g. the example does not have to be thread-safe, IMHO).

gavanderhoorn, 2015-02-26:
@tnl-woi wrote:
> I would favor a working API for the I/O interface [..]

We have a 'working API' for the I/O interface, the `SetIO` service. You're talking about a Python wrapper around that service, then?

tnl-woi, 2015-02-27:
Ah yes, you're right. I should have been more explicit, but that's what I meant ;-)

Maybe the ```SetIO``` service is a good API after all, but I haven't found any documentation about it and the Python file may have set me on the wrong path. So, summarized, I would be okay with:
* removing ```io_interface.py```
* updating ```test_io.py```, which can be used to test the I/O interface and as a simple example
* preferably adding some documentation of the ```SetIO``` interface (doesn't have to be much, possibly a reference to the example file)

gavanderhoorn, 2015-02-27:
Yes, docs for the IO interface are currently lacking, but I think that is partly because of #118.

It should be relatively easy to fix, as the ROS wiki can semi-automatically document services for us (similar to how topics and parameters are now documented for [ur_driver](http://wiki.ros.org/ur_driver)).

gavanderhoorn, 2019-11-23:
Closing, as with the deprecation of `ur_modern_driver` we'll also remove `ur_driver` in `melodic-devel` (#448), so this would no longer be an issue.

closed 2019-11-23 by gavanderhoorn