ros-industrial/universal_robot #178: suggestion: split out state reporter from driver

gavanderhoorn, 2015-03-01:
This is now possible since the driver is deserialising the state data broadcast by the UR controller itself, without the need for the `prog` URScript to be first uploaded.

I think this is a good idea for the following reasons:

 1. complexity: the state reporter can be kept relatively simple (check socket, deserialise, publish), and by removing the code from `ur_driver`, we would reduce complexity of that node. It would also increase separation-of-concerns overall.
 1. remove a spof: if the driver crashes, or is disconnected for some reason, we also loose the state reporting at the moment. By completely separating the motion (and io, and ..) interface from the state reporter, that would be no longer the case.
 1. make the state reporter usable stand-alone: I've come across some use cases where it was desirable to visualise the state of a UR but not use the rest of the driver (ie: commanding motion). While that can be done by (ab)using the `prevent_programming` parameter, a separate state reporter would be much cleaner.
 1. increase *intelligibility* (or *understandability*): though admittedly minor, I think having separate nodes (and / or even launch files) would decrease the perceived (by users) complexity of the driver as a whole.

One thing I haven't looked at is how such refactoring could potentially influence a `ros_control` compatible version of the driver (although I'm not sure that wouldn't require a rewrite in C++ anyway).

ipa-fxm, 2015-03-05:
You can already achieve a stand-alone state-reporter by running [this script](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/testRT_comm.py)....well kind of ;-)...the file currently publishs to ```joint_statesRT```

ipa-fxm, 2015-03-05:
However, as the current driver requires information about the robot state in order to determine the connection status of the robot and stuff like that, you will not be able to completely separate the ```prog```-part from the ```state-reporter```-part...

de-vri-es, 2015-03-05:
> However, as the current driver requires information about the robot state in order to determine the connection status of the robot and stuff like that, the prog-part cannot be separated from the state-reporter-part easily/completely...

But you could get that information from the state reporter instead of getting it from the arm directly. You will still need to open a connection to upload the driver script (and possibly secondary scripts for #176), but I like the idea.

If not splitting it out entirely into a different node, then splitting it more clearly in classes in their own source file without any shared global variables between the classes would already be a nice improvement to reduce complexity of the driver.

gavanderhoorn, 2015-03-05:
@ipa-fxm wrote:
> However, as the current driver requires information about the robot state in order to determine the connection status of the robot and stuff like that, the `prog`-part cannot be separated from the `state-reporter`-part easily/completely...

Could you explain this a bit? Afaik we could just connect to 30002/30003 with a proper version of `testRT_comm.py` and it *should just work* (tm)?

ipa-fxm, 2015-03-05:
Well, yes, but you cannot remove the deserialize part from the driver.py the way it currently is implemented...
E.g. the connection state is used for determining whether the robot is connected or the joint_states are used for trajectory interpolation...
We would need to subscribe from the remainder of the driver to the topics that are published by the state-reporter or something like that...

gavanderhoorn, 2015-03-05:
O, ok. I'm not suggesting we remove everything completely, just that we refactor things (as @de-vri-es suggests) in such a way that a stand-alone state reporter becomes *possible*.

I'm well aware that the current implementation of the 'rest' of the driver needs some way to close the loop.

> We would need to subscribe from the remainder of the driver to the topics that are published by the state-reporter or something like that...

Yes, that would be even nicer. Pitty that there is no such thing as nodelets in Python.

gavanderhoorn, 2018-06-25:
`ur_driver` is deprecated and we'll not be working on it any longer.

As such I'm closing this issue.

closed 2018-06-25 by gavanderhoorn