ros-industrial/universal_robot #180: ImportError: No module named cfg

tnl-woi, 2015-03-05:
In #163, an import statement was added, which breaks the driver, at least for me (using Ubuntu 14.04.2 LTS, Python 2.7.6).

```
Traceback (most recent call last):
  File "/home/wouter/catkin_ws/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 20, in <module>
    from ur_driver.cfg import URDriverConfig
ImportError: No module named cfg
```

There is a file ```ur_driver/cfg/URDriver.cfg``` added in the same PR, but I have no idea how this is supposed to work. The file seems like a Python file, but the extension is wrong, the ```ur_driver/cfg``` directory should not be accessible from the ```driver.py``` (it's not in the path, it's not in the ```src``` directory), there is no ```___init__.py```. So I'm confused...
@ipa-fxm Could you explain how this should work and why it may fail in my setup?

gavanderhoorn, 2015-03-05:
@tnl-woi: have you `catkin_make`-ed your workspace? The `dynamic_reconfigure` stuff needs a build-step, otherwise the files aren't generated, and you get that import error.

ipa-fxm, 2015-03-06:
@tnl-woi:
Could you please close this issue in case a re-compilation (```catkin_make```) solves your problem?

tnl-woi, 2015-03-09:
Ah, I didn't rebuild my workspace since I didn't know this was necessary for the driver (because it used to be Python only). I don't understand exactly what this new code does, but after ```catkin_make``` it works, so I'll close the issue. Thanks for the responses.

closed 2015-03-09 by tnl-woi

ipa-fxm, 2015-03-09:
The new ```dynamic_reconfigure``` parameter ```prevent_programming``` allows you do prevent uploading the ```prog``` immediately after it has been cancled e.g. due to interaction with the UR teaching pendant.
This allows you to use teach mode or trigger other motions from the teach pendant, without needing to kill the ur_driver.
Just execute ```rosrun rqt_reconfigure rqt_reconfigure``` and you will see the new checkbox under the ur_driver namespace...

gavanderhoorn, 2015-03-10:
@tnl-woi: you should never assume that Python projects / packages don't need a build step: it is perfectly possible for Python packages to depend on compiled extensions, which obviously do need to be build.

Also: if an update introduces new Python packages or messages/services, a build is also always necessary to update paths and generate artefacts.