ros-industrial/universal_robot #181: Configurable goal tolerances

mwojtynek, 2015-03-05:
There is a lack of configurable goal tolerances using the ur_driver. Using MoveIt! with the ur_driver to reach a defined goal will fail within 2-3 mm (cartesian) on a real robot. The tolerances of the ur_driver should be more strict for accurate pick and place applications.
The whole pick and place scenario with a lack of accuracy is described and discussed in [1]. This issue also came up several months ago in [2].

[1] https://groups.google.com/forum/#!msg/moveit-users/w5_Y8Q46kK8/FqfwIQOjQ2sJ
[2] https://groups.google.com/forum/#!msg/swri-ros-pkg-dev/2Mb4CO8anCI/uzOYiinjWuoJ

gavanderhoorn, 2015-03-05:
We could probably make the tolerances configurable by simply retrieving a ros parameter at node startup. The tolerances are in *joint order*, so a simple list should suffice.

Alternatively, we could expose the tolerances to `dynamic_reconfigure`. That would make configuration through launch files possible, as well as changing them at runtime.

ipa-fxm, 2015-03-10:
At the end of the discussion of your link [[2]](https://groups.google.com/forum/#!msg/swri-ros-pkg-dev/2Mb4CO8anCI/uzOYiinjWuoJ), you mention that you solved the problem by lowering the tolerances.
Did you change the values [here](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/driver.py#L665)? What was the value you used in the end?

ipa-fxm, 2015-03-10:
After thinking two more seconds :wink: we simply should use the toleances specified in the [FollowJointTrajectoryGoal](http://docs.ros.org/indigo/api/control_msgs/html/action/FollowJointTrajectory.html)

Given this part of the message is filled by MoveIt!...

mwojtynek, 2015-03-10:
Someone else solved the problem in [2] by lowering the tolerances. At the moment I'm looking forward for a solution. I can only suppose that this guy edited the values in Line 665. Until now I've tried values like 0.03, 0.01 and 0.005, 0.003, 0.001.

ipa-fxm, 2015-03-10:
...and did one of those values help?

ipa-fxm, 2015-03-10:
Hmm, ```rostopic echo /arm_controller/follow_joint_trajectory/goal``` shows that (at least by MoveIt!) the ```path_tolerance```, the ```goal_tolerance``` and the ```goal_time_tolerance``` are not set.... :disappointed: 

```
rostopic echo /arm_controller/follow_joint_trajectory/goal 
WARNING: no messages received and simulated time is active.
Is /clock being published?
header: 
  seq: 0
  stamp: 
    secs: 167
    nsecs: 654000000
  frame_id: ''
goal_id: 
  stamp: 
    secs: 167
    nsecs: 654000000
  id: /move_group-1-167.654000000
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs: 0
      frame_id: /world
    joint_names: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
    points: 
      - 
        positions: [3.223351353121018e-05, 0.012367118440149483, -0.0041704531744031215, 0.0003513386079783132, 2.3770304474979298e-05, 3.0239077724480978e-05]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.06933720193279741, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 0
      - 
        positions: [0.13138792550211173, -0.11621927526490755, -0.0011492119410675232, -0.003833101899406809, 0.00305069997690197, 0.0014452614954543592]
        velocities: [0.7623804325127933, -0.74630759401461, 0.017535100027459186, -0.024286237737772783, 0.017568115381336624, 0.008212703892100754]
        accelerations: [3.0458332776592663, -2.9816196852368404, 0.0700555640111118, -0.09702745235346966, 0.07018746569592048, 0.03281108190523042]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 295205567
      - 
        positions: [0.26274361749069225, -0.24480566896996458, 0.001872029292268075, -0.00801754240679193, 0.006077629649328961, 0.0028602839131842374]
        velocities: [1.243374694047064, -1.217161323664147, 0.028598188911886674, -0.03960869420169801, 0.028652034018313274, 0.013394189768857362]
        accelerations: [3.043152025040835, -2.9789949599615455, 0.0699938939696758, -0.09694203890927673, 0.07012567954118937, 0.03278219824967469]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 416854028
      - 
        positions: [0.39409930947927274, -0.3733920626750216, 0.004893270525603673, -0.012201982914177054, 0.00910455932175595, 0.004275306330914115]
        velocities: [1.528356921548985, -1.4961354309125168, 0.03515291100624007, -0.048687030728961314, 0.03521909744343608, 0.016464146117645594]
        accelerations: [2.8073281632736493, -2.748142840230947, 0.06456983686039257, -0.08942968139474698, 0.0646914099344225, 0.030241797860590683]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 510215914
      - 
        positions: [0.5254550014678533, -0.5019784563800787, 0.007914511758939272, -0.016386423421562175, 0.012131488994182942, 0.005690328748643994]
        velocities: [1.7636723685782703, -1.7264898545275948, 0.040565274343090985, -0.05618319228584598, 0.04064165126055797, 0.01899906963518539]
        accelerations: [3.0461225515658947, -2.9819028605441638, 0.07006221744379679, -0.09703666740486463, 0.07019413165578972, 0.03281419809347128]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 589836916
      - 
        positions: [0.6568106934564338, -0.6305648500851357, 0.01093575299227487, -0.0205708639289473, 0.015158418666609932, 0.007105351166373872]
        velocities: [1.76367236857827, -1.7264898545275948, 0.040565274343091, -0.05618319228584599, 0.04064165126055796, 0.018999069635185384]
[... 19 lines cut ...]
          secs: 0
          nsecs: 832779802
      - 
        positions: [1.0508777694221754, -1.0163240312003068, 0.019999476692281665, -0.03312418545110266, 0.024239207683890904, 0.011350418419563506]
        velocities: [0.7654742979855558, -0.7493362332591562, 0.017606260354013723, -0.02438479529407384, 0.017639409689639736, 0.008246032398350638]
        accelerations: [-3.080643414175148, 3.0156959391942797, -0.07085621313554397, 0.0981363570617432, -0.07098962229474018, -0.033186072305634924]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 953735134
      - 
        positions: [1.182233461410756, -1.1449104249053639, 0.023020717925617263, -0.03730862595848779, 0.027266137356317895, 0.012765440837293385]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [-3.0146007673736963, 2.9510456324251306, -0.06933720193279741, 0.09603251578690183, -0.06946775107452442, -0.03247463129888819]
        effort: []
        time_from_start: 
          secs: 1
          nsecs: 248940701
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs: 0
---
```

ipa-fxm, 2015-03-10:
One more source for inaccuracies when commanding __Cartesian__ goals, could be the performance of the IK solver!
In case you are using the KDL-Plugin, you could also try to play with the parameters in the kinematics.yaml in the moveit_config (e.g. decrease ```kinematics_solver_search_resolution``` and increase ```kinematics_solver_timeout```) or test the analytical URKinematics-Plugin

mwojtynek, 2015-03-10:
No, those values didn't help. After changing the values I got an error message, which looked, like already mentioned, somehow like this:

[WARN] [WallTime: 1425662360.897418] Trajectory time exceeded and current robot state not at goal, last point required
[WARN] [WallTime: 1425662360.898550] Current trajectory time: 12.1617338657, last point time: 12.152725802
[WARN] [WallTime: 1425662360.899532] Desired: [0.7850048455213197, -1.570023189120181, 1.5699340866921936, -1.5700101106241346, -1.5699668246168643, 0.7850814506199677]
actual: [0.7834, -1.5708, 1.5662, -1.5761, -1.5762, 0.7674]
velocity: [0.0144, 0.0042, 0.0316, 0.0541, 0.0541, 0.1593]

mwojtynek, 2015-03-10:
I've already tried to increase the search resolution of the KDL plugin, but without any results.  
You mean, I should try the URKinematics-Plugin which is already added to the ur_driver from 'Kelsey Hawkins'? I think it's slower than KDL (okay, it's analytic) but I'll give it a try!

mwojtynek, 2015-03-10:
Using the FollowJointTrajectoryGoal message with the goal_tolerance is not a bad idea to do make a workaround. As you mentioned MoveIt! doesn’t set the path and goal tolerances. But at least I’m not sure how to realize the path planning without MoveIt. 
In this case I have to calculate the trajectories by own, set parameters like time_from_start for the execution time, and of course I’ve to set the goal tolerances.  At least I’ve to send the trajectory to the action server. Am I wright?

ipa-fxm, 2015-03-10:
> I've already tried to increase the search resolution of the KDL plugin, but without any results. You mean, I should try the URKinematics-Plugin which is already added to the ur_driver from 'Kelsey Hawkins'? I think it's slower than KDL (okay, it's analytic) but I'll give it a try!

Use #184 for testing URKinematics-Plugin together with MoveIt!
URKinematics-Plugin should be __much__ faster than KDL-Plugin!

gavanderhoorn, 2015-03-12:
@mwojtynek: the messages you copy/pasted appear to actually be warnings, not errors, and the driver does appear to append an additional point. After that, the eef is not at the desired position?

mwojtynek, 2015-03-12:
@gavanderhoorn True, true it's a warning, no error. But no, the eef is not at the disired position. Sometimes the motion stops during the execution.

bluefish89, 2015-04-28:
@mwojtynek : Did you manage to get a better result reaching the point?

While playing with the tolerance values I noticed that the position reaching is getting better the more often I use the move() command. So with a small for-loop, I just use the move() command 6 times in a row (which takes only half a second) and I perfectly align with the desired position.
Maybe this little workaround helps one or another...

RonaldEnsing, 2015-09-15:
I have exactly this problem as well using MoveIt and ur_driver.

mbraun89, 2017-11-24:
I have the same problem using the UR_Driver. To realize my pick and place task I use the workaround mentioned by mwojtynek with multiple calls of move(). Is there a better possibility now?

gavanderhoorn, 2017-11-24:
No work has been done in this direction.

cgnarendiran, 2018-01-26:
@ipa-fxm You have mentioned that Moveit! doesn't set the `path_tolerance`, the `goal_tolerance` and the `goal_time_tolerance` . I can confirm this behaviour while running Cool1000 7DOF arm with `joint_trajectory_controller` spawned by controller_manager. I'm guessing it the problem of the `controller_manager` that fails to load the tolerance values off the yaml config file. Did you get any lead on this? Without the tolerance values set in the ActionGoal message, how is the trajectory execution monitor checking the violation of goal or path tolerances?

gavanderhoorn, 2019-06-29:
Closing this as it was reported against `ur_driver` (deprecated and no longer developed).

MoveIt still doesn't set goal tolerances afaik, so in case that is used, MoveIt would have to be fixed first.

If there is still interest in this feature, a PR against `ur_modern_driver` would be the fastest way to get this supported.

closed 2019-06-29 by gavanderhoorn