ros-industrial/universal_robot #182: Changing environment, adding end effector

bluefish89, 2015-03-05:
Hi,

I'm running an UR5 with your package and it's working fine. Thanks a lot for making this available!!

Now I would like put an object in the system, setting the world coordinate origin to this new object and then putting the robot in reference to this world coordinate. For example putting an table in the system, setting the world origin at on foot of the table (x=0,y=0,z=0,rx=0°,ry=0°,rz=0°) and then put the robot at some pose in reference (for example x=100,y=100,z=100,rx=45°,ry=0°,rz=90° in Euler notation).
When I put in goal poses  for the robot end effector afterwards, they should be in the same coordinate system then.

In addition to that point I want to add an end effector to the end of the arm of the UR5.

Both the object (table) and the end effector I have as an CAD-file (step or others).

Could you tell me the steps I need to take to achieve this goal? Can I use the MoveIt Setup Assistant to do all that?

Best regards
Ben

ipa-fxm, 2015-03-05:
Well, in the end you will be creating a new URDF (composition) which includes your table, the UR5 and your new end-effector.
For the UR5 part you can use the xacro macro available [here](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_description/urdf/ur5.urdf.xacro).
It can be used as can be seen [here](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_description/urdf/ur5_robot.urdf.xacro#L12).

Have a look at [these tutorials](http://wiki.ros.org/urdf/Tutorials) for more information on URDF and XACRO.
From your CAD data export the bodies as meshes and use them in the ```visual``` and ```collision``` tags.

The MoveIt!-SetupAssistant is then used to configure a moveit_config for your setup (i.e. table, UR5 and end-effector) which is different from the moveit_config for the UR5 allone provided in this repo.

bluefish89, 2015-03-06:
Thanks for your answer! Then I will work myself through the tutorials.

Just one more question to get this right: Would you recommend to edit the existing UR5 config in the MoveIt!-SetupAssistant or build a complete new one.

And one last question on top ;) Is it possible in the end that the object (table) has some dynamic parameters (for example the height of the table changes while the robot is moving)

ipa-fxm, 2015-03-06:
You definitely should create a new MoveIt! configuration package as MoveIt! configuration packages are created for a particular setup (i.e. URDF). By adding a table and/or endeffector, you are changing the setup! Of course you can use the MoveIt! SetupAssistant to create the configuration package.

In URDF you cannot dynamically change e.g. the size of a table...
In case the table is just an obstacle that should be considered for motion planning, you could use MoveIt!'s planning_scene_interface to spawn collision objects (and vary there dimension/position) during runtime by simply publishing to ```/colision_object``` topic.

bluefish89, 2015-03-06:
Thanks for all the information!

closed 2015-03-06 by bluefish89