ros-industrial/universal_robot #188: [Indigo] ros_control-enabled ur_driver

ipa-fxm, 2015-03-11:
FYI, @abubeck, @gavanderhoorn, @adolfo-rt

Based on various recent discussions, about using ros_control for the universal_robot, I'm tempted to use @davetcoleman's [boilerplate](https://github.com/davetcoleman/ros_control_boilerplate) as a start for doing this....

I'll probably start with a simple JointStateinterface first...(which would be related to #178)

We could use this issue for further discussions....

gavanderhoorn, 2015-03-11:
Good idea. I've been thinking about this myself for some time now.

I think I already mentioned this before, but my idea was to send `speedj()` statements to 30002 at a sufficient rate based on control input, then use the RT data from the controller to close the loop. Should not be too difficult.

ipa-fxm, 2015-03-11:
Yes, once we got the basic thing running, it's easy to add more modes, e.g. velocity streaming, position streaming and using the joint_trajectory_controller from ros_controllers (which would make the ur_driver internal interpolation obsolete)...

gavanderhoorn, 2015-03-11:
Yes, `prog` could go, if we feel the performance using the new approach is good enough. I've talked to a few people that experienced delays while using the secondary interface as we're discussing now, so I'll be interested to see how this works out.

gavanderhoorn, 2015-03-11:
> position streaming using `servoj` [..]

Issue is that there is no standardised Cartesian controller afaik in `ros_control`, or anywhere else in ROS.

I'd like to avoid inventing *yet another* Cartesian interface / msg / Action set.

TheDash, 2015-03-11:
+1 we've been discussing this at Clearpath as well. This would be super nice.

toliver, 2015-05-22:
We have written a c++ version of this driver ([sr_ur_arm](https://github.com/shadow-robot/sr_ur_arm)) that is implemented as a ros_control controller plugin. That allows to use it alongside a trajectory controller.

The trajectory controller does the interpolation and sets the targets (every 1ms, as we are running a 1KHz loop that controls a Shadow hand and  the UR10). The arm driver/position controller sends the targets to the arm every 8ms (125 Hz rate, which is the maximum for a UR arm).

It uses a simplified version of the `prog`.

For the moment it allows only servoj() and teach_mode() commands, but it could be easily extended to allow other commands.

adolfo-rt, 2015-05-25:
Hi @toliver, thanks for sharing your effort.

Could you describe a bit the architecture of your implementation?. I see the [sr_ur_controller](https://github.com/shadow-robot/sr_ur_arm/blob/indigo-devel/sr_ur_controller/include/sr_ur_controller/sr_ur_controller.hpp), which implements the `controller_interface::Controller` interface, but two things strike me as odd: I didn't find any hardware abstraction implementing the `hardware_interface::RobotHW` interface, and the `sr_ur_controller` embeds a [UrRobotDriver](https://github.com/shadow-robot/sr_ur_arm/blob/indigo-devel/sr_ur_controller/include/sr_ur_controller/sr_ur_controller.hpp#L62) instance, which looks as if the controller is embedding the driver that talks to hardware. I might be reading this the wrong way, but I don't see the things I normally expect in hardware abstraction and controller implementations.

toliver, 2015-05-26:
Hi @adolfo-rt, you are right that the architecture is not as one would expect. 

The `RobotHW` we are using is in https://github.com/shadow-robot/ros_ethercat/blob/indigo-devel/ros_ethercat_model/include/ros_ethercat_model/ros_ethercat.hpp. This provides a hardware abstraction for the joints that are present in the URDF (in our case 1 single URDF for arm/s + hand/s) and deals with (by loading driver plugins for them) the ethercat devices found in the specified ethernet ports.

What it doesn't do for the moment is to load non-ethercat driver plugins, although some functions for that exist in [EthercatHardware](https://github.com/shadow-robot/ros_ethercat/blob/indigo-devel/ros_ethercat_hardware/include/ros_ethercat_hardware/ethercat_hardware.h#L256) (inherited from the pr2 code base).

For the moment as you point out, the UrRobotDriver is instantiated in the sr_ur_controller, instead of being instantiated in the RobotHW. I think the main reason for this decision was time pressure in our project and the lack of a clear view of how to do it in the `RobotHW`, given our multi-robot scenario.

It would be good if you could provide us some insight into the right way to solve this.

I our case we want to have the all the robots in the same control loop, but still have the flexibility to decide if the robot will have 1 or 2 arms or no arm at all, etc.

The most straight forward way that I can think of would be to use the `loadNonEthercatDevices()` to load the arm driver as a plugin (i.e. inside the `EthercatHardware` instance that lives inside `hardware_interface::RobotHW`).

A perhaps cleaner way would be to load the arm driver as a plugin directly in the [hardware_interface::RobotHW]( https://github.com/shadow-robot/ros_ethercat/blob/indigo-devel/ros_ethercat_model/include/ros_ethercat_model/ros_ethercat.hpp), based on some configuration parameters.

This 2 options rely on havinbg a single `hardware_interface::RobotHW` instance for the whole robot. Is there a standard way in ros_control to deal with a multi-robot by creating separate `hardware_interface::RobotHW` for each one of the composing robots?

ipa-fxm, 2015-05-27:
Thanks a lot @toliver for sharing this...
Unfortunately, I haven't had a closer look yet as I don't have much time atm....hopefully I'll be able to do this in June.

But as the discussion was about using ```hardware_interface::RobotHW``` for it, I just wanted to point to a draft package [ur_driver_rc](https://github.com/ipa-fxm/universal_robot/tree/indigo_ur_driver_rc/ur_driver_rc) on my [branch](https://github.com/ipa-fxm/universal_robot/tree/indigo_ur_driver_rc).

So far, it's just a half-day effort of using @davetcoleman's [boilerplate](https://github.com/davetcoleman/ros_control_boilerplate) to be able to load a ```joint_state_controller```.
The [ur_controller_interface-part](https://github.com/ipa-fxm/universal_robot/blob/indigo_ur_driver_rc/ur_driver_rc/include/ur_driver_rc/ur_controller_interface.h) is not really usable yet, but allowed me to load and start a ```joint_state_controller``` in order to receive a ```joint_states``` message from the UR.

I will keep working on this as soon as I find the time...

gavanderhoorn, 2016-12-30:
I'm going to close this. Not because `ur_driver` is currently `ros_control` compatible, or because I don't appreciate the efforts of @ipa-fxm here, but because I feel the `ros_control` capabilities of `ur_modern_driver` already cover most of the use-cases outlined in the above comments.

To avoid duplication of effort, it would be good to focus our attention on `ur_modern_driver`.

closed 2016-12-30 by gavanderhoorn