ros-industrial/universal_robot #191: UR Firmware V3 Problems

panjia1983, 2015-03-21:
Hi,

I recently tried to utilise this package on a UR5 Robot running V3.0.16471 of their software which was released on Dec 02 2014. The result was very similar to that described in issue #16 

Here is the console output:
```
panj@panj-OptiPlex-9010:~/Workspace/APC/src/universal_robot/ur_bringup/launch$ roslaunch ur_bringup ur5_bringup.launch robot_ip:=10.10.72.135
... logging to /home/panj/.ros/log/fa75fe38-cff2-11e4-9a6b-b8ca3a8089f1/roslaunch-panj-OptiPlex-9010-25291.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://panj-OptiPlex-9010:54109/

SUMMARY
========

PARAMETERS
 * /robot_description
 * /robot_ip_address
 * /robot_reverse_port
 * /rosdistro
 * /rosversion
 * /tf2_buffer_server/buffer_size
 * /ur_driver/max_payload
 * /ur_driver/max_velocity
 * /ur_driver/min_payload

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    tf2_buffer_server (tf2_ros/buffer_server)
    ur_driver (ur_driver/driver.py)

auto-starting new master
process[master]: started with pid [25308]
ROS_MASTER_URI=http://panj-OptiPlex-9010:11311

setting /run_id to fa75fe38-cff2-11e4-9a6b-b8ca3a8089f1
process[rosout-1]: started with pid [25321]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [25333]
process[ur_driver-3]: started with pid [25351]
process[tf2_buffer_server-4]: started with pid [25352]
Setting prefix to 
[WARN] [WallTime: 1426960323.061396] No calibration offset for joint "shoulder_pan_joint"
[WARN] [WallTime: 1426960323.061693] No calibration offset for joint "shoulder_lift_joint"
[WARN] [WallTime: 1426960323.061917] No calibration offset for joint "elbow_joint"
[WARN] [WallTime: 1426960323.062105] No calibration offset for joint "wrist_1_joint"
[WARN] [WallTime: 1426960323.062239] No calibration offset for joint "wrist_2_joint"
[WARN] [WallTime: 1426960323.062417] No calibration offset for joint "wrist_3_joint"
[INFO] [WallTime: 1426960323.062626] No calibration offsets loaded from urdf
[INFO] [WallTime: 1426960323.064127] Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [WallTime: 1426960323.066008] Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [WallTime: 1426960323.104382] Programming the robot
[INFO] [WallTime: 1426960323.104587] Programming the robot at 10.10.72.135
[2015-03-22 01:52:03.508516] Halted
Waiting to program
Waiting to program
Waiting to program
Waiting to program
Waiting to program
Waiting to program

^C[tf2_buffer_server-4] killing on exit
[ur_driver-3] killing on exit
[robot_state_publisher-2] killing on exit
Traceback (most recent call last):
  File "/home/panj/Workspace/APC/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 1015, in <module>
    if __name__ == '__main__': main()
  File "/home/panj/Workspace/APC/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 985, in main
    time.sleep(1.0)
KeyboardInterrupt
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

ipa-fxm, 2015-03-22:
This is definitely related to #183 and current CB3-incompatibility of the ur_driver..
The reason why you see the ```Waiting to program``` is the second point in [my comment](https://github.com/ros-industrial/universal_robot/issues/183#issuecomment-81878857).
While your robot is actually ok and sends ```ROBOT_RUNNING``` in the new defines, i.e. 7, the driver interprets this as ```ROBOT_POWER_OFF```, which is definitely not "READY_TO_PROGRAM```!

Although I still experienced some problems with v3.1, [my branch](https://github.com/ipa-fxm/universal_robot/tree/indigo_fix_deserialize_v3) seems to work for v3.0.
Could you please test it and report back here?

ipa-fxm, 2015-03-26:
The problem with v3.0 is that it uses the new RobotMode defines but does not send a RobotMessage as a first packet that would include the major and minor version number. (This is only done since v3.1.)
Thus, I don't see any possibility to detect what version we are communicating with except for maybe introducing a ```cb_version``` (launch-file) parameter for the ur_driver...

gavanderhoorn, 2015-03-30:
@ipa-fxm: there are no pkt size differences that we might use as a clue?

Re: `cb_version`: a version selector would be an option, but I don't think we should actually call it `cb_version`, as that seems to me a bit ambiguous. UR also seems to give its control boxes 'versions', which do not follow the regular Polyscope version numbers (ie: CB3 can run `v3.0.x` but also `v3.1.x`).

ipa-fxm, 2015-04-01:
Well, the protocol change (i.e. in the defines) seemed to have happened when switching the maior version from 2 to 3. The new defines are used for both minor versions 0 and 1...

gavanderhoorn, 2015-04-01:
> when switching the maior version from 2 to 3

you mean the 'control box version', not 'polyscope version' here right?

The only thing I was worried about was the possibility of UR releasing a 3.x Polyscope that introduces breaking changes to the protocol again, which would leave us without a way of configuring the driver for it (seeing as it would still be the same control box).

The `ROBOT_MESSAGE` would allow for some auto-detection though, so perhaps it isn't such a big issue anymore.

ipa-fxm, 2015-04-01:
> The ROBOT_MESSAGE would allow for some auto-detection though, so perhaps it isn't such a big issue anymore.

Well, that's the problem for controller version 3.0 there is no such ROBOT_MESSAGE but the new defines are already used...

gavanderhoorn, 2015-04-01:
Perhaps the addition of `ROBOT_MESSAGE` was inspired by people running into the same issue as we are currently :).

de-vri-es, 2015-04-02:
Would it be an option to connect to port 30001 (primary interface) instead of 30002 (secondary interface)? There are more messages in that stream, which may contain the required version information. It would mean de-serializing more data, but most of it can still be discarded.

Or maybe even just open a connection, read messages until version information is found, close connection and then continue as usual. It could be a separate class then.

Jmeyer1292, 2015-11-18:
Is there any progress on this issue? If not, is there any documentation on the current comm protocol? @gvanderhoorn, you mentioned a wiki?

gavanderhoorn, 2015-11-18:
Afaik there is some work in [ipa-fxm/universal_robot/indigo_fix_deserialize_v3](https://github.com/ipa-fxm/universal_robot/tree/indigo_fix_deserialize_v3) that should fix this, but I'm not sure as to what its state is. You should ask @ipa-fxm.

re: wiki: UR has moved all support info to their [support pages](http://www.universal-robots.com/support/). The wiki is no longer used/updated. See [Remote Control Via TCP/IP - 16496](http://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/) for info on the protocols on the primary, secondary and 'matlab' interfaces.

matt12345678, 2016-01-28:
I'm having the same problem; has a fix been made? I installed from source on Dec. 7, 2015. I'm usually able to connect the first time, but I have trouble reconnecting after a disconnect.

shaun-edwards, 2016-01-28:
@matt12345678, sounds more like #199 is your issue.  If you aren't having this issue all the time, then it would be unlikely to be a binary protocol/parsing issue.

gavanderhoorn, 2016-12-30:
re: issues with CB3+ controllers and system versions 3.x and up: our official recommendation is to use the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) in these cases.

re: connection issue: this is / was a problem with UR's software, not this driver (ie: other, non-ROS software had the same / similar problems). Afaik, these issues have been resolved by UR.

gavanderhoorn, 2016-12-30:
Closing as this driver is not meant to be used with CB3 / v3.x system software versions.

closed 2016-12-30 by gavanderhoorn