ros-industrial/universal_robot #192: Detaching the base_link fixed to world

bluefish89, 2015-03-23:
Hi!

I would like to put the base_link of the UR5 to a certain pose using the tf- static_transform_publisher. Unfortunately for me, the base_link is fixed to the world and cannot be changed in position and orientation. Also tried it in the RVIZ-tf-tree.

Is there an easy way to remove this fix? I don't want to build a new moveit_config for it as I like to use the convenient package given by you.

Thanks and regards

gavanderhoorn, 2015-04-20:
The `urX.urdf.xacro` files do not contain a `world` link nor the `world_joint` that links `base_link` to it. Using that you should be able to position the UR anywhere you want. It would require you to make a new 'top level' urdf, as `urX.urdf.xacro` only defines a macro, it does not call it.

That would require you to at least *update* the MoveIt pkgs, but I see no other way.

ipa-fxm, 2015-04-21:
As @gavanderhoorn explained, you can use the ```urX.urdf.xacro``` files and import them into another xacro file where you can compose your scene and put the URX within it whererver you like....
See how it is done in the ```urX_robot_urdf.xacro``` files....i.e. by changing the origin in the ```world_joint``` you change the position of the robot with respect to ```world``` which also is the reference frame for gazebo.

bluefish89, 2015-04-23:
Okay, so I would have to update the MoveIt pkgs. 
Thanks @gavanderhoorn and @ipa-fxm !

My workaround right now is, that I turned the whole system in respect to world leaving the robot where it is. In my case this means: instead of putting the robot upside down (the robot is attached to the ceiling) and turned everthing else (all the collision objects) upside down.

If I would leave it that way, do you think I get problems with the kinematics as the robot may calculate with the wrong gravitational acceleration. Or is it sufficient to put this information in the UR-Controller-Pad?

Thanks and regards

bluefish89, 2015-04-24:
Okay, it seems like I do get problems. While running the robot in this setup it runs a little shaky and often I loose connection to the robot ("stopped hearing from robot"). Then my only option seems to be ctrl+c and restart.
I guess I need to change the gravity and hopefully this will solve the problem. But I can't find this option. The script-command in UR-script-language is the simple "set_gravity([0.0, 0.0, -9.82])". Could you tell me where exactly I can insert this information in the universal_robot package?

gavanderhoorn, 2015-04-24:
afaik there is no notion of gravity in the driver (or MoveIt): everything is done in joint space, and only kinematics are considered in the standard path planning in MoveIt.

Before we go any further: what version control box do you have, and which version of Polyscope is it running?

bluefish89, 2015-04-26:
User Interface:     PolyScope 1.8.11281 (Aug 07 2013)
Robot Controller: URControl 1.8.11281 (Aug 07 2013)

bluefish89, 2015-04-27:
And Controllerbox 2.

ipa-fxm, 2015-04-28:
If your robot is just located __somewhere in a scene__, rather than part of a __composed robot__ (i.e. including base), you would not need to create a new MoveIt! config necessarily.
It would be enough if you add your world to the planning scene of MoveIt! by publishing your world description to the ```/collision_object``` topic of MoveIt!

"Virtually" mounting the whole thing upside down seems to be a strange solution....
Even if you would need to create a new MoveIt! config....that's not such a big deal as it would only mean to re-run the SelfCollision-Matrix step and then use the urX_moveit_config packages as a template to fill out the rest (by hand or within the SetupAssistant).

ipa-fxm, 2015-04-28:
Furthermore, I don't think that the "stopped hearing from robot" is related to your issue, but is another issue with your connection/driver...

bluefish89, 2015-04-30:
Thanks for the information.
As you said, the  "stopped hearing from robot" seems to be another issue. Actually I found out that this could be stopped when not running RVIZ. So it seems that the computational power is not enough when dealing with the real robot and all the collision objects.

The upside down solution works fine for me at the moment. And as I can't start RVIZ with my computer, I don't recognize any difference. However, anytime soon -when I have some time - I will come back to your solution.
Thanks again!

closed 2015-04-30 by bluefish89