ros-industrial/universal_robot #193: RRTConnect as default planning algorithm for Indigo

marcoesposito1988, 2015-03-24:
Hi guys,

since upgrading to Indigo I encountered some problems with the planning performance (this seems to be a common issue: see [the moveit-users mailing list](https://groups.google.com/forum/#!topic/moveit-users/M71T-GaUNgg)).

There seem to be two possible solutions to this problem:
- set the "planning attempts" number to 1 (suggested by Stefan Kohlbrecher in the mailing list above)
- select the RRTConnect planning algorithm (I opened a bug in the OMPL tracker about the default planning algorithm: while LBKPIECE1 looks to be sensible for most robots, the URs do not seem to belong to this category)

While the first solution appears to lead to faster planning, I think that the second one should be more robust for complex scenarios. In any case I think that one of the two actions should be taken: since Indigo, so one year now, I get 2 seconds planning time out of the box for the UR5. Any of the above paths guarantees a planning time less than 0.1 seconds (on my machine).

gavanderhoorn, 2015-03-24:
On 24-3-2015 16:55, Marco Esposito wrote:
> Hi guys,
>
> since upgrading to Indigo I encountered some problems with the planning performance (this seems to be a common issue: see [the moveit-users mailing [...]

Could you be a bit more specific about what you mean with "the URs do
not seem to belong to this category". Do you have an idea why not, or is
this based on observation only at the moment? I'd like to understand the
factors that influence this, as your first suggestion seems like a
work-around for some deeper issue (also what I gathered from your
moveit-users posting).

> While the first solution appears to lead to faster planning, I think that the second one should be more robust for complex scenarios. In any case I think that one of the two actions should be taken: [...]

Agreed. For the Fanuc pkgs I'd rather rely on something like your second
suggestion.

shaun-edwards, 2015-03-24:
Does the planning algorithm selection affect planning time or just the likelihood that a solution is found?

ipa-fxm, 2015-03-24:
@marcoesposito1988 
Option1 is already on the indigo-devel branch (since #156)

gavanderhoorn, 2015-03-24:
@ipa-fxm: any more information you can share? Bugs/issues to link to? Or is that based on observation?

ipa-fxm, 2015-03-24:
Nothing beyond (the very detailed) discussion [here](https://groups.google.com/forum/#!topic/moveit-users/M71T-GaUNgg)...

marcoesposito1988, 2015-03-25:
@ipa-fxm: this was my conclusion as well, but then I debugged more and I found out that the deeper problem is the choice of the algorithm. If the parallel planning is used, then first many solutions are found in distinct threads, then Path Hybridization is performed. The problem is, that the output of LBKPIECE1 for the UR5 are paths that are 100-150 states long instead of 4-5, so the hybridization problem explodes in complexity (it is in the end the Dijkstra algorithm over the graph of all states in the paths). On the other hand the output paths of RRTConnect are 4-5 states long, so parallel planning works properly. Single-threaded planning does not make any hybridization, so this problem is less evident, but RRTConnect still behaves better. 

I would say that changing the planning algorithm is the correct way: this makes both single- and multi-threaded planning work properly. 
On a side note, single-threaded planning is still faster when there are no obstacles are in the way of the robot, so I would make that the default as well.

@gavanderhoorn: I am bugging the OMPL people as well (see [this issue](https://bitbucket.org/ompl/ompl/issue/126/wrong-algorithm-selected-by-selfconfig)) in order to find the deeper reason why the UR5 and a KUKA LWR4+ with a tool attached both create problems with LBKPIECE1, while the LWR4+ without a tool behaves correctly. Once I find an algorithmic way to determine what projection is appropriate for a given robot I will try to implement it and put it into the MoveIt Setup Assistant, where I think it belongs to.

@shaun-edwards: it affects the probability of finding a solution and the "smoothness" of the trajectory as well. LBKPIECE1 creates a lot of unnecessary states.

gavanderhoorn, 2015-03-25:
Thanks for your reply. I think I'll change the default planner then for my pkgs. Setting planning attempts to 1 is something users can easily do, if they feel planning is too slow for their use case.

gavanderhoorn, 2015-04-08:
@marcoesposito1988: any updates on this?

marcoesposito1988, 2015-04-09:
@gavanderhoorn: I can just confirm my findings, even if I am not using the UR5 at the moment. According to an OMPL developer we can force RRTConnect by "removing the "projection_evaluator" entry from ompl_planning.yaml". I can do this and make a pull request.

gavanderhoorn, 2015-04-09:
I was just wondering. I'll let @ipa-fxm handle the rest.

marcoesposito1988, 2015-05-20:
@ipa-fxm: did you have time to evaluate the proposed change? I think it's the most proper way to go. The problems with parallel planning are just a symptom of the wrong choice of the planning algorithm, and #196 fixes it.

ipa-fxm, 2015-05-21:
I'm afraid to say that I haven't yet been able to test this in more detail....
Your explanations sound all reasonable...I hope I can find the time to test this beginning of June...
Sorry!

marcoesposito1988, 2015-05-21:
@ipa-fxm: no problem, I just wanted to make sure that you don't forget about me ;)

gavanderhoorn, 2015-05-24:
@marcoesposito1988: could Hydro MoveIt also be affected? I can get some really long and weird planning results from MoveIt under Hydro if I don't set *Planning Attempts* to `1`.

Have the Indigo improvements to MoveIt been backported to Hydro?

marcoesposito1988, 2015-05-24:
@gavanderhoorn: I did have problems with hydro as well, yes.

You can directly try it out: you just have to choose RRTConnect from the drop down menu in the first tab of the moveit rviz plugin. It is actually not much more complicated than changing the planning time.

ipa-fxm, 2015-05-27:
Hmm, well, I didn't do a systematic evaluation, but when I tested your patch branch and tested indigo-devel, I'm not clear about the particular gain of this PR...

I simply tested ```roslaunch ur5_moveit_config demo.launch``` (w and w/o ```limited:=true```) and then tried to move to random_valid configurations...

Indeed, I see that MoveIt! now uses ```RRTConnect``` instead of ```LBKPIECE1```!
However, this default planner also fails to produce plans for the limited:=false sometimes...Is this the same for you?

Also, should it be possible to now set the ```PlanningAttempts``` parameter (in the rviz panel) back to a number of e.g. ```10``` (which would imply parallel planning, right?) rather than ```0``` or ```1``` and the planning results are still valid, smoother and computation is quicker than with non-parallel LBKPIECE1?

Can anyone confirm my observations?
@marcoesposito1988 how did you evaluate the performance of the new moveit_config? What values are you using for ```PlanningTime``` and ```PlanningAttempts```?

marcoesposito1988, 2015-06-01:
@ipa-fxm: as you noticed limited:=true is still required, yes. That is a different story. 

I didn't benchmark RRTConnect systematically, but with limited:=true and any value of PlanningTime and PlanningAttempts I cannot find any glitch at all.

If there were a way to explicitly specify the default planning algorithm in the yaml configuration file that would be better, but this is out of the scope of this issue and of the pull request. The purpose is instead to provide a correct default configuration that makes the robot usable out-of-the-box.

marcoesposito1988, 2015-07-15:
closed through #196

closed 2015-07-15 by marcoesposito1988

gavanderhoorn, 2015-11-25:
Just an FYI: configuring a default planner in a MoveIt configuration is now actually possible with the merge of ros-planning/moveit_ros#625. This made it into Indigo.

It would probably make sense to revert 5d0ea453 and d9875b55 (from #196), and use the newly exposed configuration interface to make RRTConnect the default planner explicitly.