ros-industrial/universal_robot #194: Re-release of package

TheDash, 2015-04-01:
Hi UR,

could this package be re-released after adding collisions for the URX arms and the fix to the free-planning joint-free mode?

This would be appreciated as we are currently having to ask people to install source software rather than debs to utilize this package to its full extent.

ipa-fxm, 2015-04-21:
The ur_driver currently has some problems (as discussed in other issues)...
Thus, the next release will need to wait a bit until things are worked out.

ipa-fxm, 2016-04-01:
I'm going to re-release the package....although the issues with the driver still exist...
Still, I think it's worth to have the updates in the `ur_description` as well as in the launch file structure available from debs...

ipa-fxm, 2016-04-01:
https://github.com/ros/rosdistro/pull/10950

closed 2016-04-01 by ipa-fxm