ros-industrial/universal_robot #195: [C API] Scons compile error - set connected() is protected

2Reeko, 2015-04-10:
Hi,

I've been doing the installation step 1 of ur_c_api_bringup and I did everything upto the compile step. Just putting the specified files into a directory on the control box and creating the symolic links for the compiler. I had an issue earlier with the scons compiler not being able to read the simple_message .h or .cpp files but it was just the character encoding that it didn't recognise. Now when I try to compile the control server it stops early with this error.

include/simple_message/socket/simple_socket.h: In member function 'void ur::URControllerInterface::sendAndReceiveMessages()':
include/simple_message/socket/simple_socket.h:253: error: 'virtual void industrial::simple_socket::SimpleSocket::setConnected(bool)' is protected
src/ur_controller_iface.cpp:51: error: within this context
scons: *** [build/test/src/ur_controller_iface.o] Error 1
scons: building terminated because of errors.

Seems that both connected_ and setconnected() are protected in the simple_socket.h. This throws an error when the compiler hits this section of ur_controller_iface specifically the sendandreceivemessages function;

  if(ur_state.sequence - latest_cmd_seq >= CMD_TIMEOUT) {
    // printf("TIMEOUT\n");
    connection->setConnected(false);

Any ideas on how to get around this? or have I just missed something I should have done?

Thanks,

Christian

gavanderhoorn, 2015-04-15:
I remember something about some methods being protected in `simple_message` and @kphawkins running into this as well some time ago. @shaun-edwards: do you remember how/if this was resolved?

@2Reeko: could you specify which particular version of the C-API driver you are attempting to use? The branch on this repository is rather old, and has not been kept up-to-date with the changes in the [gt-ros-pkg/universal_robot](https://github.com/gt-ros-pkg/universal_robot) fork.

2Reeko, 2015-04-15:
@gavanderhoorn I followed the link from http://wiki.ros.org/ur_c_api_bringup which appears to be the same as the link you posted. Not sure how to find the version 'number'? 

branch: hydro-c-api-ur10-moveit
latest commit: 8fe5153cfa
authored: kphawkins dec 5, 2014

shaun-edwards, 2015-04-16:
@kphawkins submitted a fix for UDP in ros-industrial/industrial_core#70.  Until that point, the c-api required a fork of industrial_core.  I believe his fix was not quite compatible with the original c-api, since we asked for some changes.  I would recommend looking for a newer fork as @gavanderhoorn suggests, as it will probably be compatible with the latest version of industrial_core.

kphawkins, 2015-04-16:
https://github.com/gt-ros-pkg/universal_robot/commit/1d9fc81a87ca1b032c463b32986b9147ef0fbc02

I think this commit fixes the issue. I still recommend using my fork of the
industrial_core, even though I believe this fix will make them compatible.

2Reeko, 2015-04-16:
Thank you all so much for your help

closed 2015-04-16 by 2Reeko