ros-industrial/universal_robot #197: [C API] hardware_interface/pos_vel_acc_joint_interface.h: No such file or directory

2Reeko, 2015-04-16:
HI,

I'm using indigo on 14.04 and when I try to catkin_make universal_robot I get the error:

[ 96%] Built target ur_sim_server
In file included from /home/anbot/catkin_ws/src/universal_robot/ur_ctrl_client/src/ur_robot_hw.cpp:2:0:
/home/anbot/catkin_ws/src/universal_robot/ur_ctrl_client/include/ur_ctrl_client/ur_robot_hw.h:22:60: fatal error: hardware_interface/pos_vel_acc_joint_interface.h: No such file or directory
 #include "hardware_interface/pos_vel_acc_joint_interface.h"

I had a look in hardware_interface and found posvelacc_command_interface.h instead. I was wondering if any of you knew if this is an upgraded version of pos_vel_acc_joint_interface.h or if it would be easier to just install hydro on an earlier version of ubuntu? @kphawkins @shaun-edwards @gavanderhoorn

closed 2015-04-17 by 2Reeko

gavanderhoorn, 2015-04-17:
Have you solved this? If so, it would be nice if you could tell us how, so we can either do something about it permanently, or at least document this for future users.

2Reeko, 2015-04-17:
I changed to the *hydro-devel-c-api* branch from the *hydro-c-api-ur10-moveit* branch and made a change to the *ur_controller_iface.cpp* file. This seems to work because it has *pos_vel_acc_joint_iface* in *ur_ctrl_client* instead of in the *hardware_inteface*.