ros-industrial/universal_robot #198: [C API] controller_box_type 5 not recognised

2Reeko, 2015-04-17:
Hi,
 On *Starting the controller drivers* step of the http://wiki.ros.org/ur_c_api_bringup, after bringing up the rqt and enabling plugins. When I click the 3rd button to open the interface with the robot I get the following error:

##### conf.calibration[0] : 
##### conf.calibration[1] : 
##### conf.calibration[2] : 
##### conf.calibration[3] : 
##### conf.calibration[4] : 
##### conf.calibration[5] : 

##### End of conf : 'urcontrolrc' #####

=====================================================
Error: joint configuration-files not loaded correctly
=====================================================

I tried running the ur_ctrl_server on the control box again and saw the following warning:

*******************************************************************
WARNING: Configuration is inconsistent:
Robot type is UR10, but controllerbox does not match this robot type
Please review urcontrol.conf
*******************************************************************
This lead me to find the controller_box_type is 5 for my situation and in the ur-c-api-example-1.8::configuration.h the comment next to the controller_box_type only says number 1 or 2. I'm not sure if this will solve the joint configuation error though.

 I'm now using gt-ros-pkg/universal/tree/hydro-devel-c-api with the updated ur_controller_iface.cpp and gt-ros-pkg/industrial_core  hydro-devel.

gavanderhoorn, 2015-04-17:
Could you please reformat your post? If you could put all console copy/pastes between three backticks ("```", before and after the text) that would make things a lot easier to read and understand.

gavanderhoorn, 2015-04-17:
Also: which control box do you have (CB2, CB3) and what is the version of Polyscope installed on it?

2Reeko, 2015-04-17:
I'm not in the lab with the robot at the moment but we purchased it a few weeks ago, so I would assume the latest for both.

Updated OP for readablility

gavanderhoorn, 2015-04-17:
Ok. Afaik, C-API is only supported on CB2 controllers. The latest release is `1.8.16941`, which is also the latest Polyscope version compatible with CB2 controllers. The `controller_box_type` having only the value 1 or 2 is probably a hint. I'd recommend you send UR (or your local distributor) an email to get this confirmed.

In any case, the LLSI is supposed to be the replacement of C-API for CB3 controllers, but (as the name implies) is not (directly) compatible with the C-API driver in this repository.

2Reeko, 2015-04-17:
Thanks for the information, I'll see what my distributor says.

2Reeko, 2015-04-20:
UPDATE:  Our UR10 is currently Polyscope 3.1.16916 and CB3. I will look into using the LLSI after a firmware update to 3.1.17336. 

Just a couple last questions:
Is there any documentation on the LLSI? and is the ros-industrial/universal_robot indigo branch compatible with either of these polyscope versions?

gavanderhoorn, 2015-04-20:
> Our UR10 is currently Polyscope 3.1.16916 and CB3

So no C-API for you then.

> I will look into using the LLSI after a firmware update to 3.1.17336. 

Know that that will require you to port the current C-API driver to LLSI. Afaik no one has done that yet.

> Is there any documentation on the LLSI?

You'll have to ask UR that, we're mere mortals, just as you. The archive does contain a `example_LLSI.cpp` that should show you how to use it.

> and is the ros-industrial/universal_robot indigo branch compatible with either of these polyscope versions? 

See [wiki.ros.org/universal_robot](http://wiki.ros.org/universal_robot), but no, it isn't. All 3.1.x versions currently have issues with the driver (see also #183 and #191).

2Reeko, 2015-04-20:
Thanks for the help, I'll figure something out.

Closing thread

closed 2015-04-20 by 2Reeko

gavanderhoorn, 2015-04-20:
Depending on what you intend to do, it might be worthwhile exploring a port of the C-API driver (which wouldn't be called "C-API driver" anymore then) to LLSI. I'm not sure about the feasability and / or the effort that would be involved though. It would probably be a good idea to discuss this with @kphawkins.

kphawkins, 2015-04-22:
So although it was not in my current plans a few days ago, I might be
developing a port of the C-API to LLSI in the coming months.  I'm actually
going to get access to a CB3 control box, so I should be able to write
something.

Honestly, much of what I've written will need to be discarded, so it might
be more of a rewrite than a port, but I will try and salvage as much as I
can so that it's fully compatible with ros-control.

jorritsmit1, 2015-09-18:
Has there been any progress on the porting of C-API? Or can anybody point to some information about LLSI, and where can i find the archive that contains the LLSI_example.cpp? It has been quite frustrating to get everything to work with so little information.

Regards

gavanderhoorn, 2015-09-18:
> Has there been any progress on the porting of C-API?

Just to be clear: the 'C-API' is a controller programming interface that Universal Robots provides, it is not something that we -- as ROS-Industrial -- own / support / maintain. In fact, we are not related to Universal Robots in any way.

Also: the C-API itself is not / cannot be 'ported' per se. LLSI is the successor to the C-API, and works on CB3 and up.

What @kphawkins was referring to was that the ROS-Industrial compatible driver component that *makes use of the C-API interface* would need to be updated to use the LLSI, in order for it to be usable with CB3 controllers.

> Or can anybody point to some information about LLSI, and where can i find the archive that contains the LLSI_example.cpp? It has been quite frustrating to get everything to work with so little information.

As we did not create the LLSI, nor the C-API, I can only refer you to Universal Robots. They seem to have recently migrated all their support information to a new server, and I've not been able to find anything about the LLSI on it yet.

gavanderhoorn, 2015-09-18:
If you are asking about the ROS-Industrial C-API *driver*, then I suggest you open a new issue. Note though that anything more recent than CB2 will not be able to run it.

jorritsmit1, 2015-09-18:
Ok thanks for the info, understand now. I actually have no need to get it to work with ROS, i was indeed just looking for some information on the LLSI. Also couldn't find it on the new support site, but i will send them an email. Thanks again