ros-industrial/universal_robot #199: Port 30002 hangs in ```CLOSE_WAIT``` for URs using  v3.1

ipa-fxm, 2015-04-21:
This is just a pointing issue to a discussion that evolved within issue #183 

In [comment](https://github.com/ros-industrial/universal_robot/issues/183#issuecomment-81878857) bullet 3, it is reported that the ur_driver cannot ```getConnectedRobot``` for latest UR controller versions (>3.1).
Further investigations of @ashfaqfarooqui showed that this might be a UR controller problem rather than a problem within the ur_driver only (see https://github.com/ros-industrial/universal_robot/issues/183#issuecomment-94795989)

I think we should keep the discussion within #183 to not break the flow of discussion.....
This issue is for other users looking for a solution for the problem mentioned in the issue title!

ipa-fxm, 2015-05-27:
According to Universal Robot, a solution for this bug was found and it will be fixed within a new software update soon....

I haven't yet tested it myself, I'm just forwarding this information here...
Also, I don't know the set of affected software versions and the software version with the fix...

ipa-fxm, 2015-06-02:
To give a brief update on this:
The patch/update fixing this issue is sill on hold on the UniversalRobot side...they are working on it.
Updates will be announced here: http://support.universal-robots.com/Downloads/SoftwareUpdates

As to the affected software versions:
According to UniversalRobot...
 > the bug was introduced with SW3.0.15547 (11/9/2014). It might have become more apparent with SW3.1.16828 (26/01/2015)

... when they started to use TCP_NODELAY more intensively.
Thus, software versions between [SW3.0.15547, SW3.1.17596) are affected, where I was told that SW3.1.17596 is going to be the next update fixing this issue..,

ipa-fxm, 2015-06-02:
One more thing that I would like to mention:
Even if there was a bug in the UniversalRobot software, it seems that the problem is triggered by in case a connection is made to the controller and is immediately closed again...resulting in a deadlock.

So is there something, that the ur_driver.py could do differently during connection/disconnection so that this problem does not occur?

jmfurrer, 2015-06-10:
Just to let everyone know, 3.1.17779 is out which supposedly addresses the socket issue (no mention in the release notes yet though). Will test as soon as back with the robot.

ipa-fxm, 2015-06-10:
According to my contact at UniversalRobot, the desired fix for this issue didn't make it into 3.1.17779...but the next bug-fix release should be soon(er)!

davetcoleman, 2015-06-11:
Thanks for the update!

ashfaqfarooqui, 2015-08-03:
Any idea if the bug fix made it to ursim-3.1.18024? 
The release notes states: 
>Fixed an issue that could case the Controller to hang when making a socket connection to the Controller

But I am still able to reproduce this error. did anyone else try the updated software?

ipa-fxm, 2015-08-04:
> Bugfix release 3.1.18024 should fix the connection problems (released last week)
This is my info from Universal Robots, but I haven't tested it yet! Neither in simulation nor on hardware.

Please report here (success or problems)!

clehnert1, 2015-08-20:
I'm not getting any luck. I have updated the software to 3.1.18024 but still receiving "Robot disconnected" followed by repeating "Waiting to Program" messages. If I reboot the UR5 machine it will connect successfully once, but if I kill the node and restart it, it can't reconnect. Has anyone had any luck with the update?

de-vri-es, 2015-08-20:
I have been unable to reproduce TCP communication problems so far (not using the ROS-I driver). 

@clehnert1: Can you check if the controller is still accepting connections when you encounter this problem? You can check if `nc [robot ip] 30002 | xxd` gives any output. If it produces output the controller is still accepting connections and publishing data. If it produces no output this could be the same issue as before (or a similar issue).

clehnert1, 2015-08-21:
Hi @de-vri-es, thanks for your comment. I checked if nc [robot ip] 30002 | xxd gives any output and it doesn't when the node experiences the disconnection issue. any suggestions to fix this problem if I would like to get this working by a demo in two days?

de-vri-es, 2015-08-21:
@clehnert1: Interesting, it appears to be the same or a similar problem then. One thing you can do is restart the robot controller by logging in to the robot over SSH (I think the reseller knows the password, I don't think I should post any publicly) and then restarting the controller process. I don't remember the exact commands, but I believe there are scripts in the home folder on the robot to start and stop the controller.

It's not pretty, but it should at least save a significant amount of time compared to rebooting the robot.

clehnert1, 2015-08-25:
Thanks @de-vri-es  that advice helped me get through the demo during the day. The system only froze a few times from 9-5 so I only had to use that method once or twice. Still not sure what to do about fixing the driver issue in the long term.

wair, 2015-11-10:
###### Problem:
Hi, we have problem, with unpacking robot state packet. 
###### Firmware version:
3.1.18213
###### Console output:
[INFO] [WallTime: 1447157949.435033] No calibration offsets loaded from urdf
[INFO] [WallTime: 1447157949.436608] Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [WallTime: 1447157949.439454] Bounds for Payload: [0.0, 20.0]
Disconnected.  Reconnecting
[INFO] [WallTime: 1447157949.476999] Programming the robot
[INFO] [WallTime: 1447157949.480098] Programming the robot at 10.10.10.120
Likely a syntax error:
4���������	URControlG%Aug 10 2015, 15:58:04
Exception in thread URConnection:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/baxteris/catkin_ws/src/universal_robot-indigo-devel/ur_driver/src/ur_driver/driver.py", line 279, in __run
    self.__on_packet(packet)
  File "/home/baxteris/catkin_ws/src/universal_robot-indigo-devel/ur_driver/src/ur_driver/driver.py", line 191, in __on_packet
    state = RobotState.unpack(buf)
  File "/home/wair/catkin_ws/src/universal_robot-indigo-devel/ur_driver/src/ur_driver/deserialize.py", line 324, in unpack
    raise Exception("Fatal error when unpacking RobotState packet")
Exception: Fatal error when unpacking RobotState packet

[INFO] [WallTime: 1447157950.485312] Programming the robot at 10.10.10.120
[INFO] [WallTime: 1447157951.491004] Programming the robot at 10.10.10.120

###### :
What could be done?

ashfaqfarooqui, 2015-11-10:
@wair have you tried using [this branch](https://github.com/ipa-fxm/universal_robot/tree/indigo_fix_deserialize_v3)?

wair, 2015-11-10:
@ashfaqfarooqui Thank you, it is working.

gavanderhoorn, 2016-12-30:
re: issues with CB3+ controllers and system versions 3.x and up: our official recommendation is to use the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) in these cases.

re: connection issue: this is / was a problem with UR's software, not this driver (ie: other, non-ROS software had the same / similar problems). Afaik, these issues have been resolved by UR.

gavanderhoorn, 2016-12-30:
I'm closing this issue, as it should no longer be an issue.

Please re-open if this is still a problem however.

closed 2016-12-30 by gavanderhoorn