ros-industrial/universal_robot #202: Unable to solve motion planning problem

TheDash, 2015-05-16:
Hi,

I'm having issues with the moveit configuration of the UR5 arm. I'm using the UR5Kinematics plugin with joint_limited:=false. I've disabled most of the wrongful collisions inside the .srdf. The exact bug is this:

[ INFO] [1431814422.286455475, 1541.596000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1431814422.287069467, 1541.596000000]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1431814422.287708228, 1541.596000000]: Planning attempt 1 of at most 1
[ INFO] [1431814422.303430998, 1541.609000000]: No planner specified. Using default.
[ INFO] [1431814422.310177234, 1541.613000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1431814422.310823778, 1541.613000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1431814422.312300436, 1541.613000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1431814422.327325438, 1541.624000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1431814427.305967893, 1544.174000000]: RRTConnect: Created 20 states (5 start + 15 goal)
[ INFO] [1431814427.306715531, 1544.175000000]: RRTConnect: Created 22 states (9 start + 13 goal)
[ INFO] [1431814427.314094332, 1544.178000000]: RRTConnect: Created 18 states (7 start + 11 goal)
[ INFO] [1431814427.320291821, 1544.183000000]: RRTConnect: Created 20 states (7 start + 13 goal)
[ WARN] [1431814427.320913289, 1544.183000000]: ParallelPlan::solve(): Unable to find solution by any of the threads in 5.013701 seconds
[ INFO] [1431814427.321179017, 1544.183000000]: Unable to solve the planning problem

TheDash, 2015-05-16:
It doesn't move at all. I've been using this before and had no issues, but seems just recently it's not working.

TheDash, 2015-05-17:
I've done some science regarding the limited parameters which are causing some major confusion. Although PR #184 claims to allow this to work with the unlimited setting, either I'm not having the proper version of the software (I have 1.1.5 debs), or something is wrong with my configuration, or there is a bug.

I tried running the software with the URDF model having limited:=true and limited:=false. I also checked whether or not the limited:=true/false had an affect when used in the planning_execution script.

My arm was only able to plan with UR5KinematicsPlugin when limited:=true in the URDF instantiation of the arm model. However, I could not use the free form orb and it gave a purple color in RViz. I was only able to use the update function to move it into a plannable configuration. The ideal situation is to be able to use the orb to move it anywhere and have it plan fast anywhere.

Here is the chart of science:

URDFLimited  MoveGroup Limited    CanPlan   CanUseOrb
false                     false                         No              Yes
true                      false                          Yes            No
true                      true                          Yes             No 
false                     true                          No              Yes

This lead me to conclude the movegroup limited button was having no affect as you can see from the chart. So I did indeed investigate and found out that the planning context was being used in my code, but in the ur5_moveit_config had it disabled by default. Turns out the Movegroup limited code was doing nothing anyway because load_robot_description is always set to false, and so I just removed it from my code.

That means the only conclusion can be that setting the limited arm model in URDF has all the influence of what's going on. Once upon a time I was able to plan freely with the orb and fast, but I'm not sure what happened.

ipa-fxm, 2015-05-27:
@TheDash 
The feature/fix provided in #184 is not yet released....thus you might want to use an overlay for now!

As to the confusion about the ```limited:=true```/```limited:=false``` arguments:
__This argument needs to be used consistently with all UR-related launch files you are using__
Thus, it does not make sense to use ```limited:=true``` for ur_gazebo/ur_bringup but ```limited:=false``` for moveit_config or vice versa...In case you use ```limited:=true```, you use it for all launch files. In case you use ```limited:=false```, you use it for all launch files...

Then, I'd be curious what you mean by:
> I've disabled most of the wrongful collisions inside the .srdf
Did you detect any wrong self-collision pairs?

And I don't quite understand what you mean by "ORB"? I don't know this abbreviation....Nothing from [here](http://www.abbreviations.com/ORB) seems to fit...

> However, I could not use the free form orb and it gave a purple color in RViz
When you see purple color, that means a joint_limit violation and clearly points to an invalid use of the ```limited``` argument.... or the usage of the ur_kinematics package that does not yet contain the fix from #184 

In case these hints help you figure out your problem, be so kind and close this issue!

gavanderhoorn, 2015-05-27:
@ipa-fxm wrote:
> And I don't quite understand what you mean by "ORB"?

That is probably the RViz 6D interactive marker used to drag the robot's TCP around.

ipa-fxm, 2015-05-27:
I was assuming this, but can figure out what "ORB" stands for... ;-)

gavanderhoorn, 2015-05-27:
[orb - Merriam-Webster](http://www.merriam-webster.com/dictionary/orb).

TheDash, 2015-05-27:
Sorry guys, this issue was fixed by my last comment on https://github.com/ros-industrial/universal_robot/pull/184 PLUS, I have some feedback here:

The way MoveIt works is that it determines the purple/orangeness of the manipulator in RViz by joint limits. The way to get a working, non-purple joint limit in RViz is to set limited:=true in the URDF for the UR5 arm. However, the UR5KinematicsPlugin requires the joint unlimited setup to actually solve the planning problem. This is really confusing to myself and may cause difficulties for others. E.g you'd expect to set limited:=false in the URDF because you're using an unlimited planner, but it is not the case. My suggestions would be to switch the declarations of joint limits around in the URDF so it is in line with the Kinematics Plugin.

By orb, I was talking about the RViz  interactive marker orb.

closed 2015-05-27 by TheDash