ros-industrial/universal_robot #204: README instructions not working

VictorLamoine, 2015-05-22:
```bash
$ ls src/
CMakeLists.txt  fanuc_cube_segmentation  fanuc_m10ia_ensenso_eef  fanuc_motion    fanuc_sls2-2_apps.zip  industrial_core  pcl_cloud_color_test.zip  test_my_library
fanuc           fanuc_ensenso_apps       fanuc_m10ia_kinect_eef   fanuc_pcl_apps  fermi                  my_library       test_documentation        universal_robot
```
```bash
$ rosdep check --from-paths src --ignore-src
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
All system dependencies have been satisified
ERROR[fanuc_ensenso_apps]: Cannot locate rosdep definition for [fanuc_m10ia_ensenso_eef]
	rosdep key : fanuc_m10ia_ensenso_eef
	OS name    : ubuntu
	OS version : trusty
	Data: <no data>
ERROR[fermi_sia20d_moveit_config]: Cannot locate rosdep definition for [motoman_config]
	rosdep key : motoman_config
	OS name    : ubuntu
	OS version : trusty
	Data: <no data>
ERROR[fermi_irb2400_moveit_config]: Cannot locate rosdep definition for [abb_common]
	rosdep key : abb_common
	OS name    : ubuntu
	OS version : trusty
	Data: <no data>
ERROR[test_my_library]: Cannot locate rosdep definition for [MyLibrary]
	rosdep key : MyLibrary
	OS name    : ubuntu
	OS version : trusty
	Data: <no data>
victor@desktop:~/MEGAsync/catkin_ws$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
```

```bash
$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
... logging to /home/victor/.ros/log/7f367fde-009b-11e5-87b4-10c37b6bfb43/roslaunch-desktop-2608.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://desktop:40439/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/longest_valid_segment_fraction: 0.05
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(shoulder_p...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
[... 76 lines cut ...]
 * /rviz_desktop_3159_989184372196078475/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /rviz_desktop_3159_989184372196078475/manipulator/kinematics_solver_attempts: 3
 * /rviz_desktop_3159_989184372196078475/manipulator/kinematics_solver_search_resolution: 0.005
 * /rviz_desktop_3159_989184372196078475/manipulator/kinematics_solver_timeout: 0.005

NODES
  /
    rviz_desktop_3159_989184372196078475 (rviz/rviz)

auto-starting new master
process[master]: started with pid [3173]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a6dc1db4-009b-11e5-84f1-10c37b6bfb43
process[rosout-1]: started with pid [3235]
started core service [/rosout]
process[rviz_desktop_3159_989184372196078475-2]: started with pid [3256]
Gtk-Message: Failed to load module "overlay-scrollbar"
[ INFO] [1432310422.971872561]: rviz version 1.11.7
[ INFO] [1432310422.971907585]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1432310423.477610411]: Stereo is NOT SUPPORTED
[ INFO] [1432310423.477676777]: OpenGl version: 4.4 (GLSL 4.4).
[ERROR] [1432310427.420031495]: Robot model not loaded
[ERROR] [1432310427.420290132]: Exception caught while processing action 'loadRobotModel': Character [_] is not valid as the first character in Graph Resource Name [_planning/shape_transform_cache_lookup_wait_time].  Valid characters are a-z, A-Z, / and in some cases ~.
```

ipa-fxm, 2015-05-27:
@VictorLamoine 
First of all, you need to be more detailed in your error report and more specific in what you are actually trying to do! Are you using Indigo or Hydro? Are you using debian packages or a source overlay?

Second, I don't know whether this is relevant here - as it is related to your fanuc overlay - but you should make sure that ```rosdep check --from-paths src -i``` does not return any hint on missing dependencies (note the ```-i``` which allows to ignore dependencies that you have from source in your src)!

Third, in order to use the ```roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true``` with simulation, you need to have a simulation running! Please read the "Usage with Gazebo Simulation" section of the [README](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/README.md) which tells you to run the following _three_ commands:

```roslaunch ur_gazebo ur5.launch limited:=true```

```roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true```

```roslaunch ur5_moveit_config moveit_rviz.launch config:=true```

VictorLamoine, 2015-05-27:
I thought Gazebo/MoveIt were separate instructions for doing different things, it is just me misunderstanding the instructions then.

I got the robot simulated thanks to the 3 commands:
[image]

Thanks for the fast reply

closed 2015-05-27 by VictorLamoine