ros-industrial/universal_robot #205: Duplicate Documentation on ROS Wiki and README.md

davetcoleman, 2015-06-01:
I just started using the UR5 and found the README.md on this repo first, and was left with some questions until I also found the wiki tutorial which is more complete (but mostly the same). Seems like only 1 should exist, the other having a link redirecting.

gavanderhoorn, 2015-06-02:
The tutorial was actually based on the information in the `README.md` file, so it is duplicated.

But I agree that it is something to fix. In other ros-i repositories, we always only put tutorials on the wiki, and point users to those from repository/pkg level readmes. That would seem to be most consistent.

gavanderhoorn, 2016-12-30:
Link to the tutorial mentioned by @davetcoleman: [Getting Started with a Universal Robot and ROS-Industrial](http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial).