ros-industrial/universal_robot #207: UR10 driver trajectory exceeding limits

philip-long, 2015-06-11:
Hi everybody,

I've noticed that sometimes, using the ur_driver I exceed the maximum velocity and acceleration defined 
for the robot. I am using moveit to generate these trajectories, 

As far as I can see, the following is happening:
The driver receives a set of way-points with a position,velocity, acceleration and  time_from_start. The driver then interpolates between these way-points using a cubic interpolator. The cubic polynomial is defined by boundary conditions of each way-point however in between the way-points, there is nothing to stop the sampler generating excessive velocities or accelerations. 

Kind regards
Philip

shaun-edwards, 2015-06-11:
What happens when the max velocity/acceleration is exceeded?  Does the robot fault out?

ipa-fxm, 2015-06-12:
For completness:
This issue has been reported on the ros-industrial mailinglist [here](https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/-HJeGyIEGDA)

philip-long, 2015-06-12:
In my tests, the limits that are exceeded are those defined by the trajectory which are not the same as the robots own internal limits, so there is no fault. 

Just an aside: there is nothing wrong with the interpolator, just that by using it there are cases where by using it the defined maximum velocity will be exceeded. I can send more details if you like

ipa-fxm, 2015-07-27:
@philip-long 
Of course we are very interested in any detail related to or helping to solve this issue...
Do you have any further insights on what might be causing the problem or maybe even an idea about how to solve it?

philip-long, 2015-07-28:
Ok, here's some more information and an example case with one joint moving.  I move the robot from:
q1=[-1.51128, -1.31528, -2.204, -1.46957, 1.45211, 1.56835, ] to
q2=[-1.50796, -1.31528, -0.498466, -1.46957, 1.45211, 1.56835, ] , 
with the following parameters:
max_velocity: 1.0
max_acceleration: 3.0

I have saved the  topic "/follow_joint_trajectory/goal" given from Moveit and in addition to the data sent to the robot by the driver. The robot data is taken from just after the sampling procedure  https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/driver.py#L786 . You can see that the robot velocity exceeds the maximum velocity of the trajectory (the acceleration also exceeds the limits). 

I believe this issues is happening because "/follow_joint_trajectory/goal"  gives a position, velocity, acceleration and time_from_start for each waypoint.  The cubic interpolator ensures that the robot's position, velocity, acceleration matches the trajectory at each waypoint and arrives at the waypoint at the desired "time_from_start"  as can be seen from the graphs, however there is no guarantee that the constraints with be respected between waypoints. Indeed, if the constraints are imposed, it is unlikely to reach the waypoint in the desired "time_from_start". There is an incompatibility between the cubic interpolator and the points supplied to the driver from moveit probably because moveit has optimized the waypoint to obtain a minimum traveling time. 

The way I see it there are two different approaches to solve it (neither of them particularly attractive):
1. First ensure the "/follow_joint_trajectory/goal"  points are compatible with a cubic interpolator meaning it is no longer a driver problem (it becomes a moveit or preprocessing issue)
2. Secondly, change the trajectory interpolator in the driver https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/driver.py#L584 , such that the constraints are respected but the robot can still reach the waypoint in the desired time (this is the option I set out to take, but after taking a closer look at the changes that would be required I don't think its worth it)

[image]

ipa-fxm, 2015-08-14:
Thanks for the details!
I hope, I can find some time to investigate this a bit further in September...

wasowski, 2016-09-28:
Apologies to pollute this discussion, but I was analyzing the issue with @io-sar, and it seems that this issue is not so easy to fix.  In my understanding the interpolator has the starting and ending points, but has no notion of what is an acceptable divergence from these points, or from the straight-line between them.  

You could use a heuristic like the following: if a velocity in a given direction is increasing on the segment, then you should not exceed the final velocity, and if the velocity is decreasing on a segment, then you don't go lower than the target velocity.  However it is not clear for me that it works universally.  Could not there be cases when an arm moving too slow is just as bad as when it is moving too fast in some other applications? And then doesn't one need to know a bit more about the boundary conditions? (like a constraint on a surrounding around the segment that should not be exceeded?)

I don't think that changing the interpolator itself, without the input parameters it gets, is gonna help.  You are just going to have imprecision in other places or smaller imprecision, but still imprecision.

Am I completely off?

(btw.  we we are not really going to fix this :) we just want to create a regression test, but still understanding the problem is fun)

gavanderhoorn, 2016-12-30:
We've had some discussions about this here in the lab (tud) recently (@rbbg). The interpolator in `ur_driver` (but similar approaches are used in some other packages) would certainly seem to stand some improvement, but as @wasowski wrote, this may not be that simple.

Also: the [constraint_aware_spline_smoother](http://wiki.ros.org/trajectory_filters) package in the [trajectory_filters](http://wiki.ros.org/trajectory_filters) stack could perhaps be interesting. Right now it only takes collisions into account, but it seems the infrastructure it provides could be extended to implement things like @wasowski suggests.

I won't have time to investigate this myself, so just leaving this here for future readers.

rbbg, 2016-12-30:
To add to this:

This seem to be the related to the MoveIt! issue regarding [time parameterization ](https://github.com/ros-planning/moveit/issues/160).

To summarize, the heuristic used to set the velocities and accelerations in the current MoveIt! time parameterization (Iterative Parabolic Time Parameterization - IPTP) is not playing nice with a cubic interpolator, causing the limits (mostly the acceleration, but sometimes the velocity as well) to be violated. I have done some work on improving this by fitting a spline and have seen some improvement (meaning: the violations are not as great but still there). A PR can be found [here](https://github.com/ros-planning/moveit/pull/382). 

A last step would be to perform some post processing the ensure that the limits are not violated. I haven't tested it yet but pretty sure that is just a matter of stretching the trajectory.

This is pretty much the first approach @philip-long mentioned:

> First ensure the "/follow_joint_trajectory/goal" points are compatible with a cubic interpolator meaning it is no longer a driver problem (it becomes a moveit or preprocessing issue)

This is definitely a problem that affects more than just the UR (It also happens with the `ros_controller/joint_trajectory_controller`).

gavanderhoorn, 2019-06-29:
I'm closing this. Not because it's fixed, but as this was an issue reported against `ur_driver` and that has been deprecated and is no longer being updated.

The interpolator in `ur_modern_driver` should work in a similar way, but afaik and iiuc, the type of interpolation used here and in `ur_modern_driver` is susceptible to these kind of problems.

There has been quite some work on trajectory interpolation/resampling over at `ros-planning/moveit`, so future work could consider taking a look at that and then decide what to do with `ur_modern_driver`.

closed 2019-06-29 by gavanderhoorn