ros-industrial/universal_robot #216: UR10 driver STOP_DURATION

philip-long, 2015-09-16:
In the ur_driver, if a trajectory is cancelled, the driver gives the robot 0.5 seconds to slow down gracefully starting at point0. The eventual stop point is then obtained by interpolating the trajectory 0.5 seconds into the future at point1 .

During my tests this causes an abrupt stop. I think this is because in order to stop we command the robot to move to point 1 and "arrive" with zero velocity/acceleration. In the original trajectory to move from point 0 to point 1 the robot would be travelling at a higher velocity and not necessarily decelerating, thus it can't fulfil these conditions (unless point1 coincides with the actual end point).  

In my local version I changed
point1 = sample_traj(self.traj, now - self.traj_t0 + STOP_DURATION)

to

point1 = sample_traj(self.traj, now - self.traj_t0 + STOP_DURATION - 0.3)

This means the robot has 0.5 seconds to reach a point much closer to it then before and is therefore feasible to do so while decelerating.

gavanderhoorn, 2015-09-16:
> [..] thus it can't fulfil these conditions (unless point1 coincides with the actual end point).

Or it is moving slowly enough already, if I understand your comments correctly?

de-vri-es, 2015-09-16:
Alternatively, maybe a stop deceleration could stop the robot safely regardless of the current speed of the robot. With a sane deceleration, the robot should always be able to make it.

gavanderhoorn, 2015-09-16:
Yes, I'm not too happy with relying on 'tweak factors' for these kinds of things.

philip-long, 2015-09-16:
@gavanderhoorn yes the slower the robot is moving the less noticeable the abrupt stop will be.  

@de-vri-es Yes, I agree a better solution would be to have use the maximum deceleration to calculate a feasible end point. For example from point0 we calculate point1 by maximum deceleration in 0.5 seconds. I'll have a look at this solution

de-vri-es, 2015-09-16:
Of course, if the requirement for stopping along the trajectory isn't really needed the whole procedure can be simplified further into just sending a stopj / stopl command to the robot. Otherwise it will get a bit tricky to find a point on the trajectory where the robot can stop for a given deceleration.

rkeatin3, 2017-11-21:
I'm not sure if this warrants a separate issue, but is there anything I can do to reduce the time from requesting a trajectory cancellation to the robot stopping all motion?

I was hoping to cancel the trajectory when my force reading exceeds some value, but obviously 0.5 seconds allows enough time for the force at the end effector to significantly increase.

gavanderhoorn, 2019-11-23:
I'm closing this as with the deprecation of `ur_modern_driver` we'll also remove `ur_driver` from this repository in `melodic-devel` (#448).

Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

`ur_robot_driver` uses a different approach to all of this, so it's unclear whether the issue reported in the OP is still actual.

If it is, it should be re-opened on the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn