ros-industrial/universal_robot #217: Multiple #defines when using boost and ur_kin.h

shaun-edwards, 2015-09-21:
Compiler errors result when the `ur_kin.h` file is included with boost header files.  The error is caused by redefinition of `#define` macros [here](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_kinematics/include/ur_kinematics/ur_kin.h#L47-L65).

These macros should be prefixed with a namespace and/or check for previous definition and `undef`'ed.

de-vri-es, 2015-09-21:
There's no real reason to have them as defines, but those are some very dangerously generic names for macros. Undeffing them might have weird effects if boost code uses them (though boost should know better than to use macros with names like that).

The easiest "fix" is to move them to the source file instead of the header since they're not needed in the header anyway. Better yet they can be turned into const doubles in an anonymous namespace in the source file. It should have no noticeable effect in terms of optimization with any modern compiler.

Note that the current interface prevents linking an executable that can do inverse kinematics for more than one arm type. For my own use I changed the kinematics functions to accept a struct containing the kinematic parameters instead.

de-vri-es, 2015-09-21:
Actually, after double checking, there are guarantees about using const integral types as compile time constants, but not for doubles. Although I doubt it will make a difference in practise, that could be a reason to use #defines.

Of course, c++11 constexpr would be perfect here if that was allowed.

gavanderhoorn, 2015-09-28:
Nothing more than a +1 from me, this is a really nasty issue, just ran into it myself.

ipa-fxm, 2015-10-01:
I (most likely) will be able to test this on Monday...

PR #218 (2015-10-30): ur_kinematics: Move #defines to constants in source file.
To fix #217
closed 2015-10-30 by ipa-fxm