ros-industrial/universal_robot #223: Fuse values on CB3 safety control board

jeff-o, 2015-11-25:
Well, this isn't so much of an issue as a question.  But I figured this was the most efficient way to contact a large number of people in possession of a UR arm, so here goes.

I am working on a custom install of a UR5 arm.  It is brand new, and has the latest "CB3" Safety Control Board.  During commissioning one of the fuses, F801, blew after only a few hours of use.  Based on extensive back-and-forth with the supplier, I have reason to suspect that the fuse value on our board was incorrect.  Unfortunately, UR will not reveal the correct value so that we can replace it.  They want us to buy a new board.  So, I turn to you fine folks for assistance.

If you have a CB3 board, could you please check the marking on fuse F801?  It is the one nestled between the Ground input spade connector and the large white power resistors.  On my board it is marked "1" which I assume is 1 Amp.  For reference, there are two other fuses on my board, F1001 and F1002, which are marked "5A".  These are close to one of the four-position terminal blocks.

Thanks very much for your assistance, and sorry for misusing the issue tracker. :)

gavanderhoorn, 2015-11-25:
@jeff-o: I'd suggest to also send a message to the ROS-Industrial [mailing list](https://groups.google.com/forum/#!forum/swri-ros-pkg-dev). That might reach additional people.

jeff-o, 2015-11-25:
By luck, a second UR arrived at our facility just today.  I was able to verify that F801 is indeed supposed to be 5A.  If you have the chance, I'd still like someone else to verify.  Thanks, all.

jeff-o, 2015-12-01:
I have verification that fuse F801 is supposed to be 5A.  If you've been using your robot for a while without issues, your fuse is probably fine.  But if you've got a newish one, and it's either DOA out of the box or dies soon after using it a little bit, check that fuse first.

closed 2015-12-01 by jeff-o