ros-industrial/universal_robot #225: Missing dependency to run UR robots Gazebo simulation (ur_gazebo)

chcorbato, 2016-01-06:
The `moveit_simple_controller_manager` need to be installed, as identified here:
https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/f4jF8s8_U40

gavanderhoorn, 2016-01-06:
This is probably actually a missing `run_depend` in the manifests of the `*_moveit_config` packages ([here](https://github.com/ros-industrial/universal_robot/blob/a43a0334ba3652f08e0fc937358a49985562eaee/ur5_moveit_config/package.xml#L17-L24) for the UR5 fi).

`MoveItSimpleControllerManager` is used in [ur5_moveit_controller_manager.launch.xml](https://github.com/ros-industrial/universal_robot/blob/e34e5bde886576b291b204bc701dab142dd8cf1d/ur5_moveit_config/launch/ur5_moveit_controller_manager.launch.xml#L5), but is never declared to be a dependency.

ipa-fxm, 2016-02-20:
fixed in #236

PR #236 (2016-04-01): add missing dependency for moveit_simple_controller_manager
`moveit_plugins` brings in the dependency to `moveit_simple_controller_manager` (see [rosdistro](https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L5236-L5246))

fixes #225
closed 2016-04-01 by ipa-fxm