ros-industrial/universal_robot #229: ur_driver unable to connect

varunagrawal, 2016-01-30:
I am trying to connect to a UR5 robot. Running the launch files gives me the message "Robot connected" so I am assuming that ROS is able to connect to my UR.
But when I try to run `rosrun ur_driver test_move.py` it gets stuck on "Waiting for server..." and never continues.

Any ideas what could be the problem?
Running on Ubuntu 14.04, ROS Indigo.

gavanderhoorn, 2016-01-30:
We're going to need more information: what is the version of the main system software running on your controller?

Also, please see if #191 or #199 seem related.

Finally, if this is a serialisation issue, you might want to test the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) package.

varunagrawal, 2016-02-01:
So this is a CB2 controller running version 1.8.16841 (recently updated, previously version 1.7 at the time the issue was logged).
The links provided are for the CB3 controller. Any hints or tips?

de-vri-es, 2016-02-01:
The error is actually the `test_move.py` action client waiting for the action server to become available [1]. So either the action server was not running or the client is using the wrong name.

[1] https://github.com/ros-industrial/universal_robot/blob/f5dc6c93af442744dbdeae69abf2441194c76492/ur_driver/test_move.py#L95

varunagrawal, 2016-02-01:
How do I figure out if the action server is running? `rostopic list` does not give me that information.
Also I don't understand what you mean by "using the wrong name". The node seems to be assigning itself the name "test_move".
I'm sorry if these are stupid questions. I'm quite new to ROS in general and don't have prior experience with robotic interfaces. I really appreciate the time and help.

varunagrawal, 2016-02-02:
Hey!
I got the test_move.py to work. Seems like the update did the trick. Thank you for all your help!

As a side, can you give me more details about Trajectory Planning for the UR using this driver?

ashfaqfarooqui, 2016-02-02:
Trajectory planning is done using [MoveIt](http://moveit.ros.org/), You can find some good tutorials [here](http://aeswiki.datasys.swri.edu/rositraining/indigo/).

I suggest you take that discussion on the mailing list, and mark this as resolved.

closed 2016-02-02 by varunagrawal

hariharan382, 2020-10-31:
> Trajectory planning is done using [MoveIt](http://moveit.ros.org/), You can find some good tutorials [here](http://aeswiki.datasys.swri.edu/rositraining/indigo/).
> 
> I suggest you take that discussion on the mailing list, and mark this as resolved.

getting unable to reach the webpage
 and also tried to update but not solved yet for me

gavanderhoorn, 2020-10-31:
Please do not randomly spam issues from almost 5 years ago.