ros-industrial/universal_robot #230: Cannot compile universal robot: URDriverConfig.h error 127

VictorLamoine, 2016-02-02:
```
: No such file or directory
make[2]: *** [/home/dell/catkin_workspace/devel/include/ur_driver/URDriverConfig.h] Error 127
make[1]: *** [universal_robot/ur_driver/CMakeFiles/ur_driver_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

VictorLamoine, 2016-02-02:
The problem is due the the git configuration handling of CRLF; switch to
```git config --global core.autocrlf input```

Then re-clone the repository and run `catkin_make` again.

A detailed explanation of the problem is explained [here](http://askubuntu.com/a/372691/73160).

closed 2016-02-02 by VictorLamoine

gavanderhoorn, 2016-02-02:
Just to clarify: this is a client-side configuration option. The sources in the repository are ok.