ros-industrial/universal_robot #231: Failure to execute trajectories with gazebo simulation and missing UR model

scottpaulin, 2016-02-14:
I found a few problems while following the readme instructions for running a UR5 with gazebo simulation. These are possibly related so I have grouped them in this issue. 

Software:
Clean install of Ubuntu 14.04. ROS indigo. ros-indigo-moveit-full and universal_robot packages from apt-get. 

Missing simple controller manager package:
Moveit could not load the controller manager, so I installed ros-indigo-moveit-simple-controller-manager. Maybe this was not the correct thing to do as I had problems with the controller failing. 

Controller failure:
When trying to execute a path planned in RVIZ there is a warning about trajectory timestamping in the terminal I launched the Gazebo model from:
`[ WARN] [1455491440.537708925, 88.760000000]: Dropping first 1 trajectory point(s) out of 37, as they occur before the current time.
First valid point will be reached in 0.125s.`

The robot then starts to move, but stops shortly after. There is then a warning about path tolerances being exceeded in the terminal which started ur5_moveit_planning_execution.launch:
`[ WARN] [1455491441.141944913, 89.361000000]: Controller  failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1455491441.142105548, 89.361000000]: Controller handle  reports status ABORTED`

Missing UR model:
When I booted up RVIZ, the UR5 model seemed to be missing some parts.

UR5:
[image]

Similar issue with the UR10:
[image]

Same controller problems when using the UR10.

scottpaulin, 2016-02-15:
"Clean install of 14.04" should be clean install of 12.04 upgraded to 14.04. All software was installed after the upgrade to 14.04.

gavanderhoorn, 2016-02-15:
Just to make sure: this is with Gazebo 2 or 5?

gavanderhoorn, 2016-02-15:
@scottpaulin wrote:
> Missing simple controller manager package:
Moveit could not load the controller manager, so I installed ros-indigo-moveit-simple-controller-manager. 

Yes, this is a know problem (see #225). The `readme` doesn't mention it unfortunately, but it is mentioned on [ros/wiki/ur_gazebo](http://wiki.ros.org/ur_gazebo).

> Missing UR model:
When I booted up RVIZ, the UR5 model seemed to be missing some parts.

I just checked the meshes in the released `ur_description` and they appear fine: both in Meshlab as well as in RViz.

Can you tell us a bit about your graphics hardware (nvidia/ati/intel/something else)? Installed drivers? Is this a VM or a actual hw? Gazebo/RViz are known to have issues with virtualised 3D hw, and what your screenshots show (missing faces) is typical of such problems.

> Controller failure:
When trying to execute a path planned in RVIZ there is a warning about trajectory timestamping in the terminal I launched the Gazebo model from:
```
[ WARN] [1455491440.537708925, 88.760000000]: Dropping first 1 trajectory point(s) out of 37, as they occur before the current time. First valid point will be reached in 0.125s.
```

> The robot then starts to move, but stops shortly after. There is then a warning about path tolerances being exceeded in the terminal which started ur5_moveit_planning_execution.launch:
```
[ WARN] [1455491441.141944913, 89.361000000]: Controller failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1455491441.142105548, 89.361000000]: Controller handle reports status ABORTED
```

(Virtual) controllers not able to properly control joints and / or time overruns of traj segments could both point to performance issues. Gazebo is quite CPU heavy (especially in a VM).

ipa-fxm, 2016-02-15:
You should use the sources from https://github.com/ros-industrial/universal_robot (`indigo-devel` branch)
Quite a number of fixes haven't been released into debian packages yet (see https://github.com/ros-industrial/universal_robot/compare/indigo...indigo-devel?expand=1)
Especially this includes the switch from effort-controlled hardware-interfaces in Gazebo to position-controlled hardware-interface....This should fix the controller abortion issue!
The meshes definitely seem to be a graphics (driver) problem on your setup.

scottpaulin, 2016-02-15:
The meshes now look fine with the apt-get package and the source install. Maybe some updates installed overnight. Nvidia graphics card with proprietary drivers. 

I am running on hardware, CPU is an i7-3930K CPU @ 3.20GHz. Using Gazebo 2.2.3.

The joint angles seem to become very large around the time that the controller fails. I have attached a small amount of output from the joint_states topic showing the shoulder_pan_joint moving to a position of ~400rad. 

[joint_states.txt](https://github.com/ros-industrial/universal_robot/files/131584/joint_states.txt)

Everything appears to work fine when I boot up a UR simulator (not Gazebo) in a VM and use the "MoveIt! with real Hardware" instructions.

ipa-fxm, 2016-02-20:
the dependency to the `moveit_simple_controller_manager` is added in #236

gavanderhoorn, 2016-08-31:
@scottpaulin: are you still running into issues with the simulation? Your latest comment seems to suggest this would be more likely an issue with Gazebo, `gazebo_ros_control` or even `ros_control` itself, rather than the packages in this repository (we're just users of those packages here).

scottpaulin, 2016-08-31:
Thanks for following up on this @gavanderhoorn!
 
It was still a problem when I last tested it around May on Indigo binaries using the latest source for this Universal Robot package. The robot would follow a path for a bit and then go completely unstable. Unfortunately my computer isn't in a good state to test it right now, but I will give it another go in the next couple weeks. I got the errors by following the instructions on the Readme to launch the UR5 in simulation. 

> Your latest comment seems to suggest this would be more likely an issue with Gazebo, gazebo_ros_control or even ros_control itself

Yea it could definitely be that! Everything seems to work really well when using the Universal Robot simulator instead of Gazebo.

gavanderhoorn, 2016-09-20:
@scottpaulin: any updates?

scottpaulin, 2016-09-21:
Sorry for the delay. I am having problems getting the UR5 up in simulation so that I can test whether this issue is resolved:

> roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true

Results in:

> [ INFO] [1474458804.225606945, 17.137000000]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1474458809.230629084, 22.137000000]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1474458814.235448197, 27.137000000]: Action client not connected: /follow_joint_trajectory
[ INFO] [1474458814.321328128, 27.223000000]: Returned 0 controllers in list
[ INFO] [1474458814.336289677, 27.238000000]: Trajectory execution is managing controllers

Using kinetic binaries, source install of Moveit, and source install of Universal Robot package. I just tried reinstalling ROS and Moveit but getting the same issues. Is there something obvious I have missed here?

Everything is fine when using the simulator supplied by Universal Robot (which appears to this package as a real robot).

scottpaulin, 2016-09-29:
Yea I am copying the commands from the Readme and am starting Gazebo. Does using the UR5 with Gazebo to follow paths work on your end @gavanderhoorn ? I am trying to figure out whether this whole issue could be something isolated to my machine.

gavanderhoorn, 2017-01-02:
@scottpaulin: I just noticed that your latest comments are actually based on your experiences with the packages on Kinetic. The packages are not released for that version of ROS, nor have we verified that they work (I'm pretty sure they don't all work, see #281 fi).

I've just again verified that following the instructions in the readme for using the pkgs with Gazebo and `indigo-devel` works under Indigo.

If you are no longer using Indigo, then I'd like to ask you whether we can close this issue and continue debugging this in #281, as that issue is solely about running the simulation(s) on Kinetic (and your issue started out with something rather different).

scottpaulin, 2017-01-04:
> I've just again verified that following the instructions in the readme for using the pkgs with Gazebo and indigo-devel works under Indigo.

Awesome!

> If you are no longer using Indigo, then I'd like to ask you whether we can close this issue and continue debugging this in #281,

Sure!

closed 2017-01-04 by scottpaulin