ros-industrial/universal_robot #234: [Suggestion] use args for ur_description

ThomasTimm, 2016-02-19:
Right now we have two robot.urdf.xacro files for each robot; one with limited joint movements and another without.

As Xacro has support for arg (just like launch files), I think it would make more sense to just have one file, and then pass the limited arg from the launch file to the xacro file.

The same could be applied for the prefix tag, instead of ppl having to make a copy of the robot.urdf.xacro file if they are using two arm and thus needs to specify a prefix.

ThomasTimm, 2016-02-19:
This will only be possible once we move to jade, as proper conditional evaluation is needed.

gavanderhoorn, 2016-02-19:
Yeah. And that is exactly why the current implementation is as it is right now ..

ThomasTimm, 2016-02-19:
The issue is with handling empty strings; for some reason xacro is really bad at handling that. The 'limited' parameter will always be either true or false, so that can still be changed now.

gavanderhoorn, 2018-06-25:
Related: #144.

Yaml with substitution args or even the new [yaml support in xacro](http://wiki.ros.org/xacro#YAML_support) could be approaches to consider here.

Configuring the xacro macro from a yaml file completely would have my preference.

ipa-led, 2018-07-11:
Didn't finish it yet. Will try to push it this week.

ipa-led, 2018-07-26:
Proposition of solution: https://github.com/ros-industrial/universal_robot/pull/371