ros-industrial/universal_robot #238: ERROR: cannot launch node of type [moveit_ros_move_group/move_group]: moveit_ros_move_group

Lmaths, 2016-04-24:
When I ran the following command:
`roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true`  
I got an error:
**ERROR: cannot launch node of type [moveit_ros_move_group/move_group]: moveit_ros_move_group**

shaun-edwards, 2016-04-25:
Have you downloaded MoveIt?  If you run `roscd moveit_ros_move_group`, does it find the package?

Lmaths, 2016-05-03:
`# roscd  moveit_ros_move_group`
`roscd: No such package/stack 'moveit_ros_move_group'`
But I have installed the following packages:
```
ros-indigo-moveit-core   
ros-indigo-moveit-ros-planning  
ros-indigo-moveit-msgs  
ros-indigo-moveit-ros-perception  
```
And `source /opt/ros/indigo/setup.bash` , but it did not work.

Lmaths, 2016-05-03:
I should install the full package `ros-indigo-moveit-full`, now it woks! Thank you very much. @shaun-edwards

closed 2016-05-03 by Lmaths