ros-industrial/universal_robot #240: Release into Kinetic

jonbinney, 2016-05-05:
Are there any other packages currently blocking release?

gavanderhoorn, 2016-08-30:
A kinetic release of MoveIt is close, but still not out. So this remains blocked by that at least unfortunately (ignoring the Gazebo issues @pgorczak mentioned).

gavanderhoorn, 2016-12-30:
As MoveIt is now available on Kinetic, this is no longer blocked (or at least: not by MoveIt).

SimonBirrell, 2016-12-31:
Any idea when we can expect a kinetic release?

SimonBirrell, 2017-01-01:
For those who are also waiting for the Kinetic release, here's how I got it working:

1. Install universal_robot repo into your catkin workspace
2. Add the patch from https://github.com/ros-industrial/universal_robot/pull/254/files
(This is basically adding a few new lines to ur_kinematics/CMakeLists.txt)
3. catkin_make 
4. Install ros-control (this solves the problem mentioned above by @pgorczak). 

    sudo apt-get install ros-kinetic-ros-control ros-kinetic-ros-controllers

See http://wiki.ros.org/ros_control for more details.

This seemed to do it for me. A deb install for kinetic would still be nice though.

shaun-edwards, 2017-01-02:
@SimonBirrell, thanks for reporting.  With MoveIt released in kinetic, we can begin releasing ROS-Industrial packages.  Unfortunately, we don't have a timeline for these releases.  We'd welcome an PRs that get's us closer to a release.

SimonBirrell, 2017-01-02:
The existing pull #254 would get you to where I got.

mikaelarguedas, 2017-05-25:
Now that #254 got merged. Is there any other blocker for a kinetic release?

davetcoleman, 2017-07-31:
I built this locally today and I don't see any blockers for a kinetic release, +1

ipa-rmb, 2017-08-01:
+1 for releasing

AndyZe, 2017-08-01:
It builds for me (on Kinetic), but I get this error when running `roslaunch ur_gazebo ur5.launch.` 
`joint_state_controller` is installed by sudo apt install:

> [joint_state_controller_spawner-7] process has died [pid 11838, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/controller_manager spawn joint_state_controller __name:=joint_state_controller_spawner __log:=/home/nrgadmin/.ros/log/7f8c5b46-76f7-11e7-9269-d07e35252268/joint_state_controller_spawner-7.log].
log file: /home/nrgadmin/.ros/log/7f8c5b46-76f7-11e7-9269-d07e35252268/joint_state_controller_spawner-7*.log
[arm_controller_spawner-8] process has died [pid 11852, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/controller_manager spawn arm_controller __name:=arm_controller_spawner __log:=/home/nrgadmin/.ros/log/7f8c5b46-76f7-11e7-9269-d07e35252268/arm_controller_spawner-8.log].
log file: /home/nrgadmin/.ros/log/7f8c5b46-76f7-11e7-9269-d07e35252268/arm_controller_spawner-8*.log

ipa-fxm, 2017-08-02:
I could trigger the release (I guess, I still have the according access rights :wink:)
However, I have lost proper overview of the functional state of the repo...as it is been a while I was using UR arms...

Could the more active contributors help me get relevant PRs merged and issues closed? @gavanderhoorn 
I will then trigger both tracks - indigo and kinetic

In particular, I see the following relevant:
 - #265, #268, #302 (I guess the proposed changes should also be applied to ur3?)
 - #308 
 - I will also get indigo updates to kinetic - see https://github.com/ros-industrial/universal_robot/compare/kinetic-devel...indigo-devel?expand=1

jacknlliu, 2017-08-02:
I think the most import problem is `ur_driver`, fewer people use this repo with latest real UR robot then fewer people use this repository, and then fewer people contribute for this repository. We should know that `ur_modern_driver` is out of the ros industrial. If this repository couldn't work with the latest UR robot, why we do these?

ipa-fxm, 2017-08-02:
`ur_driver` has known issues with latest UR HW...(I'm not going to start a discussion about this here)
I just would like to get PRs merged that are already proposed and "approved by discussion" as well as get respective issues closed in order to include them in the next release...

ipa-fxm, 2017-08-04:
I'm going to do a release of `kinetic-devel` in order to get the `rosdep` keys out...
another release with missing features/merged PRs can easily be done any time soon...

ipa-fxm, 2017-08-04:
releases are:
 - indigo: https://github.com/ros/rosdistro/pull/15686
 - kinetic:

jonbinney, 2017-08-04:
Thanks @ipa-fxm !

gavanderhoorn, 2017-09-13:
Closing: the pkgs have been released into Kinetic.

tnx @ipa-fxm for taking the time to do that.

closed 2017-09-13 by gavanderhoorn