ros-industrial/universal_robot #241: Using the UR robot, the arm fails in MoveIt on a real robot due to protective stop

TheDash, 2016-05-07:
The arm enters protective stop mode during the execution phase, after it has found a plan. It gets stuck in the same spot always, which is the wrist 2 link getting close to the forearm and then the arm triggers protective stop. 

My thoughts are that the collision boxes in the URDF need to be bigger for the wrist and forearm so MoveIt and the Kinematic planner don't plan the wrist to be so close. Barring that, I might have to make it a fixed link and remove its joints so it doesn't plan for the wrist at all, but the rest of the robot would. This is the worst case scenario.

I'm looking for some input on how to solve this here, as it is quite annoying and problematic for users. @gavanderhoorn

shaun-edwards, 2016-05-08:
I've had lot of problems with protective stops on the UR for high speed motion.  Could that be your issue?

Do you believe the arm is doing it's own self collision checking and triggering a protective stop?  I didn't think the UR did this?

If you are correct, I would recommend increasing the link padding in MoveIt and not altering the collision models.

gavanderhoorn, 2016-05-08:
I think I agree with @shaun-edwards: the urdf should model the arm itself as faithfully as possible. Making MoveIt avoid planning paths too tight seems like something that should be configured on the MoveIt side (ie: in the configuration pkgs). Increasing link padding is probably a good approach.

Imo users are expected to make their own MoveIt configuration packages, and by not changing the urdf we don't force our decision on all users (who may not all be using MoveIt, but would like to use the urdf).

Would that work for you @TheDash?

PS: I'm flattered, but did you have any particular reason for @mentioning me? :)

TheDash, 2016-05-08:
@shaun-edwards 

could high speed be my issue? Is there a way to turn it to low speed? There's only one spot where it continually gets stuck - the wrist link 2 and the forearm, where the wrist turns slightly too close to the forearm and it triggers it's hardware e-stop.

Okay, I did not know about the padding in MoveIt, i can try that out and report back.

@gavanderhoorn I remembered you helping me out last time I had an issue with the UR5's so I figured you were still a maintainer. Plus you're fairly attentive with the issues!

Thanks guys, I'll try the padding and see how it is

TheDash, 2016-05-09:
Do you have a tutorial on how to add padding? I cannot find anything out there

shaun-edwards, 2016-05-09:
@TheDash, you've called my bluff.  I know the padding should be settable from a yaml file.  See [here](https://github.com/ros-planning/moveit_ros/issues/341) for more info.

The max speed can be reduced in the `joint_limits.yaml`.  For some background on high speed motion issues, see [here](https://github.com/ros-industrial/universal_robot/issues/89)

I believe this problem is really a UR issue, in that the robot should not be e-stopping because there isn't a collision.  Everything we have proposed is really working around the UR issue.  However, this may be the best approach given our lack of control/access to the UR.

TheDash, 2016-05-09:
I figured out how to add padding. Just add a .yaml file to the planning_context.launch file in the moveit_config package. The .yaml should have a default_robot_scale and default_robot_padding parameter defined which sets things.

Once I added a 7cm padding, it planned and executed without issue. But, the robot was never in collision even with 0 padding. So there's that too.

VictorLamoine, 2016-06-02:
We experience the same issue, the robot is  stopping because of rapid changes in target speed (reported on the teach pendant).

We overrided safety limits to their maximum but still experience this.
Is it the ROS driver sending this to the UR or the UR triggering a e-stop?

We tried different speeds, lowering the maximum speeds in the MoveIt config but nothing works, the robot always stops after a few centimeters.

gavanderhoorn, 2016-06-02:
If the TP shows the error in the log on the robot controller it's the controller raising the error. The driver doesn't control that.

But seeing as you are using the modern driver, I'd report it [there](https://github.com/ThomasTimm/ur_modern_driver/issues).

TheDash, 2016-06-02:
@VictorLamoine 

I had no issues after I set padding in the yaml file. You've tried that right? You can also sex maximum velocities there too.

VictorLamoine, 2016-06-02:
Yes we tried to set the padding and lowering the maximum velocities.

TheDash, 2016-06-02:
Are you using the UR5 Kinematics plugin?

VictorLamoine, 2016-06-02:
No we are using KDL (because we're not using the original TCP and did not re-generate a IKFast solution)

TheDash, 2016-06-02:
That's your issue. I have never had success with the KDL. I suggest switching over to using the URKinematics, it's quite easy to do and there are no requirements.

gavanderhoorn, 2016-06-02:
On 06/02/2016 06:33 PM, Devon Ash wrote:
> That's your issue. I have never had success with the KDL. I suggest switching over to using the URKinematics, it's quite easy to do and there are no requirements.

I'd actually recommend using trac_ik in that case. Solve rates have been 
quite acceptable for us with that plugin.

TheDash, 2016-06-02:
> I'd actually recommend using trac_ik in that case. Solve rates have been
> quite acceptable for us with that plugin.

have you compared the two? I'm all for performance improvements. If this can perform better than UR Kinematics, I'd be willing to change. At the moment I believe we're getting 0.9-1.5 seconds between pressing the plan and execute button and having it reach the goal.

gavanderhoorn, 2016-06-02:
For the kind of single-query work that MoveIt typically does the performance difference between IKFast / `ur_kinematics` and trac_ik are minimal - they don't contribute that much to the total planning time. Other workloads (like the 10K+ queries something like Descartes does) every microsecond counts, so there it might make sense to use something as fast as IKFast / ur_kinematics.

An important advantage of trac_ik is that it inherits the flexibility of KDL (no need for code generation, solver can be configured by using a URDF, etc), while greatly improving solve rates. 

I've not seen anyone do any performance comparisons between `ur_kinematics` and trac_ik. That would be interesting.

TheDash, 2016-06-21:
thanks, this ticket is solved.

closed 2016-06-21 by TheDash