ros-industrial/universal_robot #242: Record the UR5 states and trajectory planning

Chunting, 2016-06-02:
Hi,
My question is not an issue but I have no idea where to put it...
Now I am working on a UR5 for my compliant force (Impedance Control) experiment, and I decide to use universal_robot project (using ur_modern_driver instead of ur_driver).
My first question is
 **How can I record all the UR5 states( position, velocity, tool position, tool force...) into a txt or csv file at rate 125Hz?**
I noticed that in ur_modern_driver the two topics has been subscribed, however, I did not receive the exact message, other saying , speed_sub_->data.c_str() is null.
```
speed_sub_ = nh_.subscribe("ur_driver/joint_speed", 1,
					&RosWrapper::speedInterface, this);
			urscript_sub_ = nh_.subscribe("ur_driver/URScript", 1,
					&RosWrapper::urscriptInterface, this);
```
My secode question is 
**Can someone please give me details on how to provide point by point Cartesian Trajectories to a UR5?**
So anyone can give me some help? If with an example how to use universal_robot on UR5, I will appreciate it well.

Thanks,
Chunting

gavanderhoorn, 2016-06-02:
> My question is not an issue but I have no idea where to put it...

Could you please post this to [ROS Answers](http://answers.ros.org)? The issue tracker is really more for issues with the packages.

gavanderhoorn, 2016-06-02:
I just noticed you have already posted this to ROS Answers ([here](http://answers.ros.org/question/235873/how-can-i-record-all-the-ur5-states-position-velocity-tool-position-tool-force-into-a-txt-or-csv-file-at-rate-125hz/)).

Closing this one.

closed 2016-06-02 by gavanderhoorn