ros-industrial/universal_robot #243: error in roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch

xsmxsm, 2016-06-02:
when i run:
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch

it has a error

NODES
  /
    move_group (moveit_ros_move_group/move_group)

auto-starting new master
process[master]: started with pid [3221]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a7d3816e-2889-11e6-8e7b-0025ab7527e4
process[rosout-1]: started with pid [3234]
started core service [/rosout]
process[move_group-2]: started with pid [3251]
 [ERROR] [1464848238.815035261]: Robot model not loaded
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [_] is not valid as the first character in Graph Resource Name [_planning/shape_transform_cache_lookup_wait_time].  Valid characters are a-z, A-Z, / and in some cases ~.
[move_group-2] process has died [pid 3251, exit code -6, cmd /opt/ros/indigo/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/xsm/.ros/log/a7d3816e-2889-11e6-8e7b-0025ab7527e4/move_group-2.log].
log file: /home/xsm/.ros/log/a7d3816e-2889-11e6-8e7b-0025ab7527e4/move_group-2*.log

gavanderhoorn, 2016-06-02:
Have you made sure to follow the steps in the [README.md](https://github.com/ros-industrial/universal_robot/blob/f68bc6c9bfa65a7dca6da711516923d07bd9938a/README.md) file?

Starting `ur5_moveit_planning_execution.launch` in isolation is not going to work.

xsmxsm, 2016-06-02:
I download the UR package from github ,and then catkin_make , then  run roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch,  and there is a error.How can I do it?

gavanderhoorn, 2016-06-02:
Closely read the `README.md` file. Follow steps under *Usage with real Hardware* or *Usage with Gazebo Simulation*. You *cannot* run `ur5_moveit_config ur5_moveit_planning_execution.launch` by itself. You *need* the other launch files.

xsmxsm, 2016-06-02:
Can I use UR in rviz if I don't have a real robot?

gavanderhoorn, 2016-06-02:
If you're asking whether you can simulate the robot, then yes. Use the steps under *Usage with Gazebo* in the `README.md` file.

xsmxsm, 2016-06-02:
When I run 
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true

it doesn't go on,just like this:

NODES
  /
    move_group (moveit_ros_move_group/move_group)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[move_group-1]: started with pid [8711]
[ INFO] [1464853437.999182092]: Loading robot model 'ur5'...
[ INFO] [1464853437.999260481]: No root joint specified. Assuming fixed joint
[ INFO] [1464853438.228069529]: Loading robot model 'ur5'...
[ INFO] [1464853438.228103692]: No root joint specified. Assuming fixed joint
[ INFO] [1464853438.392266284]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1464853438.395754538]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] [1464853438.395808624]: Starting scene monitor
[ INFO] [1464853438.398491919]: Listening to '/planning_scene'
[ INFO] [1464853438.398527079]: Starting world geometry monitor
[ INFO] [1464853438.401032788]: Listening to '/collision_object' using message notifier with target frame '/world '
[ INFO] [1464853438.403639562]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1464853438.415001278]: Listening to '/attached_collision_object' for attached collision objects
Context monitors started.
[ INFO] [1464853438.506866089]: Initializing OMPL interface using ROS parameters
[ INFO] [1464853438.531769422]: Using planning interface 'OMPL'
[ INFO] [1464853438.605366534]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1464853438.606010974]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1464853438.606516827]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1464853438.607035952]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1464853438.607496075]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1464853438.607962498]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1464853438.608027242]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1464853438.608055315]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1464853438.608079528]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1464853438.608105112]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1464853438.608128762]: Using planning request adapter 'Fix Start State Path Constraints'

xsmxsm, 2016-06-02:
I tried every one in README.md file , and it always show:

[ERROR] [1464854127.877957364]: Robot model not loaded
[ERROR] [1464854127.878405277]: Exception caught while processing action 'loadRobotModel': Character [_] is not valid as the first character in Graph Resource Name [_planning/shape_transform_cache_lookup_wait_time].  Valid characters are a-z, A-Z, / and in some cases ~.

gavanderhoorn, 2016-06-02:
Open **three** terminals.

In **terminal 1**:
```
roslaunch ur_gazebo ur5.launch limited:=true
```

in **terminal 2**:
```
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true
```

in **terminal 3**:
```
roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```

And do this **in that order** (first **1**, then **2**, then **3**).

This should start a RViz instance with the MoveIt motion planning plugin that you can use to interact with the robot.

Make sure you have installed `ros-$DISTRO-moveit-full`, or you'll probably run into problems with missing plugins.

xsmxsm, 2016-06-02:
ok ,thank you very much. I'll try it.

xsmxsm, 2016-06-02:
It successed, thank you very much!

closed 2016-06-02 by gavanderhoorn

hariharan382, 2020-10-31:
Waiting for server...
for me it is just showing this forever

gavanderhoorn, 2020-10-31:
Please do not randomly spam issues from almost 5 years ago.