ros-industrial/universal_robot #246: [Question] Potential Fitting Error

wtokumaru, 2016-06-14:
In "universal_robot/ur_driver/src/ur_driver/driver.py" the "interp_cubic" does not appear to have correct coefficients (assuming spline interpolation: https://en.wikipedia.org/wiki/Spline_interpolation). The acceleration should not be dependent on either position. Is something off by a factor of two or am crazy?

BrettHemes, 2016-06-23:
Yes, these are indeed correct despite looking strange...

For cubic interpolation we have:
q(t) = d*t^3 + c*t^2 + b*t + a
qdot(t) = 3*d*t^2 + 2*c*t + b
...

To do the fit, we are provided with q(0), q(T), qdot(0), and qdot(T)

We can easily see that:
q(0) = p0 = a
dqot(0) = v0 = b

We are left with two unknowns (c and d) and two equations (q(T) and qdot(T)).
Writing this system out we get:
q(T) = d*T^3 + c*T^2 + v0*T + p0 = p1
qdot(T) = 3*d*T^2 + 2*c*T + v0 = v1

Solving these for c and d, you will end up with the result in the code.

Hope this helps

gavanderhoorn, 2016-08-31:
Thanks @BrettHemes for that. @wtokumaru: does that help to clear things up?

If so, could you please close this issue?

wtokumaru, 2016-08-31:
Yes, thank you.

closed 2016-08-31 by wtokumaru