ros-industrial/universal_robot #250: Moveit planning execution launch files do not set controller_joint_names parameter

BrettHemes, 2016-06-23:
Not sure if this is missing for a reason but the other robot packages in the ros-industrial set this parameter.  Fixed via PR #249

gavanderhoorn, 2016-06-24:
The reason these packages don't set the `controller_joint_names` parameter is because they don't make use of the shared infrastructure provided by the `industrial_robot_client` nodes, hence they have no need for it.

Ultimately this is because the `universal_robot` stack is one of the oldest (if not the oldest: first commit (eaff5f37) was March 2012) in ROS-Industrial. It was created before any of the conventions of the other packages were set up.

gavanderhoorn, 2017-09-13:
Closing as the associated PR was also closed.

Thanks @BrettHemes for reporting this :+1:

closed 2017-09-13 by gavanderhoorn