ros-industrial/universal_robot #257: UR robot disconnected

hustfzt2, 2016-08-12:
[image]

when I interrupt the terminator "roslaunch ur_bringup ur3_bringup.launch robot_ip:=192.168.1.1", and restart it. it happens "robot disconnected". and I must power off the ur controller and power on the controller, the "robot connected" is printed in the terminator.

however, how to deal with the problem "robot disconnected" without powering off the ur controller?

gavanderhoorn, 2016-08-30:
A bit late perhaps, but if you're still struggling with this:

> when I interrupt the terminator `roslaunch ur_bringup ur3_bringup.launch robot_ip:=192.168.1.1`

you seem to be using a UR3. If you're running a Polyscope version later than v3.0.15547, you might be running into some compatibility issues (see also the [universal_robot](http://wiki.ros.org/universal_robot) wiki page).

If you have a more recent Polyscope version, please see if the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) package works for you. It is a bw-compatible replacement for `ur_driver`.

gavanderhoorn, 2016-09-17:
@hustfzt2: could you please either update this issue with more details so we can help you, or close it if you've been able to get the `ur_modern_driver` to work?

gavanderhoorn, 2016-09-20:
Closing because of inactivity.

@hustfzt2: please re-open if you can provide more details.

closed 2016-09-20 by gavanderhoorn

hustfzt2, 2016-09-23:
sorry to reply later, I will try ur_modern_driver package