ros-industrial/universal_robot #258: How to control the ur_5 or ur_10 in gazebo?

libing64, 2016-08-25:
Is there a interface for controlling the ur_5 or ur_10 robot in gazebo? Maybe something like the package for simulation of quadrotor (http://wiki.ros.org/hector_quadrotor).

gavanderhoorn, 2016-08-30:
> Is there a interface for controlling the ur_5 or ur_10 robot in gazebo?

I'm not sure I understand what you mean with that. I've never worked with the `hector` packages, so I don't know if there is something similar.

The `ur_gazebo` packages provide -- besides the Gazebo compatible launch files -- a `ros_control` compatible `position_controllers/JointTrajectoryController` configuration, which exposes the same interface as the driver for the real system (in either `ur_driver` or `ur_modern_driver`).

That Action server interface allows you to send `JointTrajectoryActionGoal` messages (using a simple Action client), which allows for basic control of the simulated arm.

libing64, 2016-08-30:
@gavanderhoorn Great thanks.
It seems the interface I want for controlling the universal robot, I will have a try.

libing64, 2016-08-30:
@gavanderhoorn Now I find the the launch file which uses the 'joint_state_controller' and I can control the universal robot by publishing the joint state message.

However,  I don't find the launch file for JointTrajectoryController, can you provide an example or the link for the launch file?

gavanderhoorn, 2016-08-31:
> @gavanderhoorn Now I find the the launch file which uses the 'joint_state_controller' and I can control the universal robot by publishing the joint state message.

I'm not sure I understand you: the `joint_state_controller` only publishes `JointState` messages. It does not accept command inputs.

> However, I don't find the launch file for JointTrajectoryController, can you provide an example or the link for the launch file?

You should be able to use the launch files in [ur_gazebo/launch](https://github.com/ros-industrial/universal_robot/tree/indigo-devel/ur_gazebo/launch). Those will load all the required controllers and setup the Gazebo instance for you.

gavanderhoorn, 2016-09-17:
@libing64: I'm guessing you got things to work. Could you close this issue if that is the case?

closed 2016-09-20 by libing64