ros-industrial/universal_robot #259: ROS Jade compatibility

Josque, 2016-08-30:
Hi, I want to use this package with ROS jade.
so can you create a branch Jade and make sure that it works with ROS Jade and ROS-Industrial Jade.

gavanderhoorn, 2016-08-30:
> so can you create a branch Jade

we typically only create release-specific branches if there is an actual need for those (ie: if there are source or binary level incompatibilities between artefacts in this repository and the ROS release).

Afaik that is not the case for the packages in this repository and ROS Jade, so that should allow you to use the Indigo packages on a Jade installation.

If you do find that something doesn't work, please report that back to us and we'll be happy to resolve the issue.

> make sure that it works with ROS Jade and ROS-Industrial Jade.

As I wrote earlier, afaik, these packages should work on Jade, they just haven't been released as binaries. That means you'll have to build them from source in your workspace, as opposed to installing them using `apt-get`. Functionally though, this results in identical setups.

odysseusPolytropos, 2016-09-17:
Hey guys,
I'm not 100% sure it's Jade-related, but there seems to be some API changes in /opt/ros/jade/lib/libmoveit_common_planning_interface_objects.so

When I try to start
`roslaunch ur5_moveit_config moveit_rviz.launch config:=true`
RViz crashes directly. When I start the same thing with Indigo, RViz comes up without a problem.

Regards,
odysseus

Log:
```
... logging to /home/rosuser/.ros/log/43ef8b8e-7cf3-11e6-afa3-448a5b057f41/roslaunch-rosuser-ThinkCentre-E73-12571.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rosuser-ThinkCentre-E73:41011/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.16
 * /rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509/manipulator/kinematics_solver_attempts: 3
 * /rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509/manipulator/kinematics_solver_search_resolution: 0.005
 * /rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509/manipulator/kinematics_solver_timeout: 0.005

NODES
  /
    rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509-1]: started with pid [12589]
[ INFO] [1474132214.894199741]: rviz version 1.11.10
[ INFO] [1474132214.894368147]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1474132215.193676703, 1289.916000000]: Stereo is NOT SUPPORTED
[ INFO] [1474132215.194038609, 1289.917000000]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1474132219.713890704, 1294.436000000]: Loading robot model 'ur5'...
[ INFO] [1474132219.714027713, 1294.436000000]: No root joint specified. Assuming fixed joint
[ INFO] [1474132219.996312905, 1294.715000000]: Loading robot model 'ur5'...
[ INFO] [1474132219.996368610, 1294.715000000]: No root joint specified. Assuming fixed joint
[ INFO] [1474132220.185905669, 1294.907000000]: Starting scene monitor
[ INFO] [1474132220.191813137, 1294.913000000]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1474132220.193314135, 1294.914000000]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1474132225.215572505, 1299.927000000]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-jade-moveit-ros-planning-0.6.6/planning_scene_monitor/src/planning_scene_monitor.cpp:461
[ INFO] [1474132226.170979566, 1300.881000000]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
**/opt/ros/jade/lib/rviz/rviz: symbol lookup error: /opt/ros/jade/lib/libmoveit_common_planning_interface_objects.so: undefined symbol: _ZN22planning_scene_monitor19CurrentStateMonitorC1ERKN5boost10shared_ptrIKN6moveit4core10RobotModelEEERKNS2_IN2tf11TransformerEEEN3ros10NodeHandleE**
[rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509-1] process has died [pid 12589, exit code 127, cmd /opt/ros/jade/lib/rviz/rviz -d /home/rosuser/auto_project/clean_workspace/src/universal_robot/ur5_moveit_config/launch/moveit.rviz __name:=rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509 __log:=/home/rosuser/.ros/log/43ef8b8e-7cf3-11e6-afa3-448a5b057f41/rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509-1.log].
log file: /home/rosuser/.ros/log/43ef8b8e-7cf3-11e6-afa3-448a5b057f41/rviz_rosuser_ThinkCentre_E73_12571_7096638322878502509-1*.log
```

gavanderhoorn, 2016-09-17:
@odysseusPolytropos: can you run other MoveIt configurations successfully under Jade?

The MoveIt configurations in this repository do not contain any binary components that are linked to anything. They are just collections of configuration and launch files. I'm not saying it's impossible that there is something not working, but it would make more sense to me if there was something with MoveIt itself not entirely ok.

MoveIt has only recently been released in Jade, so it could well be that there are still some things that need fixing.

Also, just to make sure: do you have anything built from sources in a mix with `deb` installs? Any outstanding ROS `deb` updates?

odysseusPolytropos, 2016-09-17:
@gavanderhoorn Indeed updating some MoveIt/RViz/RViz-Plugin packages for ros-jade solved the problem. Thank you very much for your quick and helpful reply!

gavanderhoorn, 2016-09-17:
No problem. Good to hear you got it to work.

@Josque: have you been able to get things to work as well?

gavanderhoorn, 2016-09-20:
Closing this.

@Josque: if there is anything we can help you with, please submit a new issue or re-open this one.

closed 2016-09-20 by gavanderhoorn