ros-industrial/universal_robot #260: how to install ur10 image

yqy0805, 2016-08-30:
Since "The ur_driver package has been successfully used with system versions ranging from v1.5.7849 to v3.0.15547. ", I tried to change the version of PolyScope to v3.0.15547, but failed, and the  PolyScope doesnt work any more.  
I know to re-install the controller system may solve the problem.  I found the image file in official website of universal robot, Image SW 3.3.1.237. (http://www.universal-robots.com/download/?option=23942#section18982). My question is how to install it? how to make an install usb key??
my robot is ur10 cb3.

gavanderhoorn, 2016-08-30:
@yqy0805 wrote:
> Since "The ur_driver package has been successfully used with system versions ranging from v1.5.7849 to v3.0.15547. ", I tried to change the version of PolyScope to v3.0.15547, but failed

It doesn't help you very much, but that same page then continues:

> Users with newer system versions are encouraged to test the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) package. 

The `ur_modern_driver` package is completely bw-compatible with `ur_driver`, and works with the latest Polyscope versions. No need to downgrade anything.

As to your problem: I think the best course of action for you would be to contact your local UR support office. The page you link to on `universal-robots.com` states:

> ONLY TO BE USED IN EMERGENCY

> Image file with default settings. 

> NOTE: Each robot arm has custom data. This in not included in this image file.

The loss of that *custom data* includes calibration and other robot specific data, which means you'll have to restore that as well, or recreate it. I don't feel we are in the best position to help you with that, I'm afraid.

BrettHemes, 2016-08-30:
There is a manual on the same site that has instructions on how to do this
(safely).

yqy0805, 2016-08-31:
We have solved the problem. I will test the ur_modern_driver for Polyscope V3.3.

gavanderhoorn, 2016-08-31:
Thanks for reporting back @yqy0805, good to hear that you got the issue resolved.

Triggered by your ticket I've given more visibility (turned it into an *admonition*) to the note about using `ur_modern_driver` with newer Polyscope versions.

gavanderhoorn, 2016-08-31:
As you've solved your original issue, I'll close this one.

Do please open a new one if you run into anything else (but please report issues with `ur_modern_driver` on the issue tracker of that repository).

closed 2016-08-31 by gavanderhoorn