ros-industrial/universal_robot #262: ur_diver - waiting on server...

shaharlevy29, 2016-09-20:
I know this issue has been here already, but i've tried every solution suggested and i still cant get it to work.

Im following the tutorials for the Universal_robot package. I've established a connection 
recieved:
[INFO] [WallTime: 1474374444.432846] Robot connected
[2016-09-20 15:27:24.438572] Out: hello

but the test_move.py is stuck on waiting for server...
I tried using the one in ur_modern_driver with no luck.
My UR version is 3.218744
The controller is fine, i can send it commands using SocketTest.

Any suggestions?

miguelprada, 2016-09-20:
Which driver are you using? The [`ur_driver`](https://github.com/ros-industrial/universal_robot) or the [`ur_modern_driver`](https://github.com/ThomasTimm/ur_modern_driver)? I'm guessing the latter, since the `ur_driver` is, AFAIK, not compatible with UR v3.X.

Could you provide a more detailed description of the exact steps you're following?

gavanderhoorn, 2016-09-20:
There is currently no point in trying to use `ur_driver` with anything over 1.8.x, so let's ignore that.

If you can test with the `ur_modern_driver` and run into the same problems, then please report this to @ThomasTimm over at `ur_modern_driver`.

gavanderhoorn, 2016-09-20:
@miguelprada: from the exceptions it would seem this is actually `ur_driver`.

miguelprada, 2016-09-20:
@gavanderhoorn There's more detail in the original post now. Guess @shaharlevy29 edited after I replied.

Close this?

gavanderhoorn, 2016-09-20:
Let's give @shaharlevy29 an opportunity to respond first.

But it would seem like this is a `WONTFIX` for now, at least for `ur_driver`.

gavanderhoorn, 2016-09-21:
@shaharlevy29: :bell: ?

gavanderhoorn, 2016-12-30:
Closing because of inactivity.

@shaharlevy29: if you have any updates, or things are still not working for you, please re-open.

closed 2016-12-30 by gavanderhoorn

philipsgithub, 2017-05-31:
I had the same errors (no attribute q_actual and fatal error when unpacking RobotState).

If somebody still has similar issues:

1. Make sure the robot is actively sending its status via tcp when a client is connected.
--> Test this with TestSocket3.0 or similar. Just connect to ur robot ip and port 30002. There should be a steady stream of incomming packets.

2. If your robot is sending its status, there is hope:) No lets come to the errors. The errors mentioned above (q_actual and unpacking problems) are not the real problem. The real problem is: "RobotStateRT has wrong length: 1060". Do a text search with this error in your universal_robot package and you will find it in deserializeRT.py. The length 1060 is unknown and belongs to the new protocol version 3.4 (see the xlxs-file at https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/ for more information).

3. Quickest fix (on your own responsibility): Just add to the if-statements in the unpack method in deserializeRT.py:   
elif plen == 1060: 
            return RobotStateRT_V30.unpack(buf)

Notes: You don't get the OutputStates and the ProgramState (new in v3.4) when you do it this way.

htugal, 2017-07-18:
I have the same problem while testing for UR3, yet can't figure out how to solve the problem. Any suggestion ?

gavanderhoorn, 2017-07-18:
Don't use `ur_driver` with UR3s. Please see the `readme` in the repository root and the *Compatibility* section on the [wiki/universal_robot](http://wiki.ros.org/universal_robot#Compatibility) page.

htugal, 2017-07-18:
Actually I am using with ur_modern_driver

gavanderhoorn, 2017-07-18:
Then you should not post your issue on this tracker.

Please post it on the issue tracker of `ur_modern_driver`.

ProfJust, 2020-01-24:
Got the same Problems described above. `ur_modern_driver `is deprecated by now.
Since I use  `UniversalRobots/Universal_Robots_ROS_Driver` including the `fmauch/universal_robot` - repository which is replacing  `ros-industrial/universal_robot ` this problem ist solved.

gavanderhoorn, 2020-01-24:
> Since I use `UniversalRobots/Universal_Robots_ROS_Driver` including the `fmauch/universal_robot` - repository which is replacing `ros-industrial/universal_robot ` this problem ist solved.

I'm glad you got things to work, but some clarification: `fmauch/universal_robot` will be merged back into `ros-industrial/universal_robot` so this is only a temporary situation.