ros-industrial/universal_robot #263: Deprecation warnings

scottpaulin, 2016-09-21:
Not sure if anyone else has noticed these warnings or how long they have been around for. I am using ros kinetic binaries with a source install of Moveit on Ubuntu 16.04.
> roslaunch ur_gazebo ur5.launch

Gives these warnings: 

> xacro: Traditional processing is deprecated. Switch to --inorder processing!
> To check for compatibility of your document, use option --check-order.
> For more infos, see http://wiki.ros.org/xacro#Processing_Order
> deprecated: xacro tags should be prepended with 'xacro' xml namespace.
> Use the following script to fix incorrect usage:
>         find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
> when processing file: /home/scott/test_ur_workspace/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> 
> redefining global property: pi
> when processing file: /home/scott/test_ur_workspace/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> xacro.py is deprecated; please use xacro instead

And this:
> roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true 

Gives these warnings:
> [ WARN] [1474452011.197816005, 88.442000000]: 
> Deprecation warning: parameter 'allowed_execution_duration_scaling' moved into namespace 'trajectory_execution'.
> Please, adjust file trajectory_execution.launch.xml!
> [ WARN] [1474452011.198551320, 88.443000000]: 
> Deprecation warning: parameter 'allowed_goal_duration_margin' moved into namespace 'trajectory_execution'.
> Please, adjust file trajectory_execution.launch.xml!

scottpaulin, 2016-09-21:
It looks like #251 fixes the xacro warnings

gavanderhoorn, 2016-09-21:
This is a known issue, and #251 does indeed fix it.

gavanderhoorn, 2016-09-21:
I'm going to close this as a *duplicate*, although technically it's only a duplicate of a non-existent issue (the one that made @davetcoleman submit that PR).

gavanderhoorn, 2016-09-21:
But thanks for reporting @scottpaulin.

gavanderhoorn, 2016-09-22:
Re-opening as the second warning is actually not related to the xacros, but to the recent changes to the trajectory execution monitor parameters in MoveIt.

gavanderhoorn, 2017-09-13:
Namespace issue was fixed with the merge of #285.

closed 2017-09-13 by gavanderhoorn