ros-industrial/universal_robot #269: Kinetic Branch fails to build

atomoclast, 2016-11-02:
Hello all, 

I am currently trying to build the Kinetic branch on a fresh install of 16.04 and a fresh install for ROS Kinetic 1.12.5 and I keep experiencing failures. I was wondering if anyone had any similar experiences or if it's still an issue that needs to be sorted out. 

it seems that there's an issue with the ur_moveit_plugin package in the build. I'll explore further, it may be a versioning issue. I just wanted to ask.

Below is an excerpt of the errors i'm getting: 

`   ^
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp: In member function ‘void ur_kinematics::URKinematicsPlugin::getRandomConfiguration(const KDL::JntArray&, const std::vector<double>&, KDL::JntArray&, bool) const’:
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:140:56: error: ‘state_’ was not declared in this scope
   joint_model_group_->getVariableRandomPositionsNearBy(state_->getRandomNumberGenerator(), values, near, consistency_limits_mimic);
                                                        ^
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp: In member function ‘virtual bool ur_kinematics::URKinematicsPlugin::initialize(const string&, const string&, const string&, const string&, double)’:
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:187:3: error: ‘robot_model_’ was not declared in this scope
   robot_model_.reset(new robot_model::RobotModel(urdf_model, srdf));
   ^
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:317:3: error: ‘state_’ was not declared in this scope
   state_.reset(new robot_state::RobotState(robot_model_));
   ^
/home/andrew/catkin_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:318:3: error: ‘state_2_’ was not declared in this scope
   state_2_.reset(new robot_state::RobotState(robot_model_));
   ^
universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/build.make:62: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o' failed
make[2]: *** [universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:1677: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/all' failed
make[1]: *** [universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
universal_robot/ur_kinematics/CMakeFiles/ur10_moveit_plugin.dir/build.make:62: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur10_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o' failed
make[2]: *** [universal_robot/ur_kinematics/CMakeFiles/ur10_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:1640: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur10_moveit_plugin.dir/all' failed
make[1]: *** [universal_robot/ur_kinematics/CMakeFiles/ur10_moveit_plugin.dir/all] Error 2
universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/build.make:62: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o' failed
make[2]: *** [universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:1714: recipe for target 'universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/all' failed
make[1]: *** [universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
`

gavanderhoorn, 2016-11-02:
I created the `kinetic-devel` branch some time ago, but it's essentially a copy of `indigo-devel` at this point.

It won't build on kinetic. See #254, #261 and #264 for related PRs.

I'll actually remove the `kinetic-devel` branch, as it's going to confuse users.

gavanderhoorn, 2016-11-02:
Sorry for the confusion.

Just to let you know: this is on our radar, and we are working on it.

gavanderhoorn, 2016-11-11:
I'm going to leave this issue open, as there is no other issue tracking the breakage of these packages on Kinetic.

SimonBirrell, 2016-12-26:
Any news on this?

WangJIankun1992, 2017-03-19:
I have successfully using universal_robot on Ubuntu 16.04 and ROS Kinetic.

gavanderhoorn, 2017-03-19:
Thanks for the report. The `kinetic-devel` branch was actually patched up to work on Kinetic some time after @atomoclast reported this.

gavanderhoorn, 2017-03-19:
Closing, as this should no longer be an issue.

closed 2017-03-19 by gavanderhoorn

marcoojer, 2018-11-23:
Hi, I'm experiencing the same error, with the current kinetic-devel branch, ubuntu 16.04 , any idea of how to fix it?