ros-industrial/universal_robot #270: Support for UR3 version 3.3.2.266?

mat-l, 2016-11-11:
Hello all,
we tried to control a real UR3 with this "ur_modern_driver".
The resulting Error message was more or less always the same. ROS could not load the robot controller. There war only one exception with the ur3_ros_control.launch. With this the robot reacted when pressing against it resulting in oversteering. could be a problem caused by the PID controller.

Furthermore, the start, using the following command "~/catkin_ws_ur3/src/universal_robot-indigo-devel/ur_modern_driver/launch$ roslaunch ur3_bringup_joint_limited.launch robot_ip:=172.31.1.150 [reverse_port:=REVERSE_PORT]" is not working.
   [image]
Repost from:
https://github.com/ThomasTimm/ur_modern_driver/blob/master/launch/ur3_bringup_joint_limited.launch

gavanderhoorn, 2016-11-11:
Thanks for the report.

I think you meant to say: *Repost from: ThomasTimm/ur_modern_driver#75*?

In any case: this would seem to be a problem in `ur_modern_driver`, as it might not have support for the protocol changes that UR made in `v3.3.2.266`. UR changes the binary format in the primary, secondary and realtime interfaces from time to time, which means that anyone interfacing with the controller using those protocols will need to catch-up to those changes.

gavanderhoorn, 2016-11-11:
As this is a duplicate of ThomasTimm/ur_modern_driver#75, I'm going to close this one.

gavanderhoorn, 2016-11-11:
If it turns out this is not a problem in `ur_modern_driver`, please re-open this issue. But do also clarify the OP with some more information.

closed 2016-11-11 by gavanderhoorn