ros-industrial/universal_robot #271: URxx.urdf.xacro files need specific link names.

atomoclast, 2016-11-11:
Hello, I am in the process of using MoveIt to configure two robots in a system to work and plan in tandem with each other. I have more information on the issue here: 

http://answers.ros.org/question/247919/xacros-macros-and-using-properties-for-moveit-with-the-universal-robots/

The basic issue, is I'm trying to make an instance of a UR10 and an instance of a UR5 robot in a single urdf file for me to use MoveIt. I'm getting odd problems because in the urXX.urdf.xacro file for each respective robot we have declarations for joints that use generic names like:

```
<property name="ur5_d1" value="0.089159" />
  <property name="ur5_a2" value="-0.42500" />
  <property name="ur5_a3" value="-0.39225" />
  <property name="ur5_d4" value="0.10915" />
  <property name="ur5_d5" value="0.09465" />
  <property name="ur5_d6" value="0.0823" />

  <!-- Arbitrary offsets for shoulder/elbow joints -->
  <property name="shoulder_offset" value="0.13585" />  <!-- measured from model -->
  <property name="elbow_offset" value="-0.1197" /> <!-- measured from model -->

  <!-- link lengths used in model -->
  <property name="shoulder_height" value="${ur5_d1}" />
  <property name="upper_arm_length" value="${-ur5_a2}" />
  <property name="forearm_length" value="${-ur5_a3}" />
  <property name="wrist_1_length" value="${ur5_d4 - elbow_offset - shoulder_offset}" />
  <property name="wrist_2_length" value="${ur5_d5}" />
  <property name="wrist_3_length" value="${ur5_d6}" /> 
```

So what's happening is, the second loaded instance of a UR is overwriting the link length property names and so on because they share the same name.

I propose that instead of "shoulder_height" and things like "upper_arm_length" we should use: "urXX_shouldler_height" or "urXX_upper_arm_length" to make multiple robot instances in a single xacro possible.

gavanderhoorn, 2016-12-30:
Two possible approaches:

 1. prefix properties with the robot model / type, as suggested by the OP
 1. use xacros new namespacing support to push all constants down into their own namespace (this is supported in Indigo as well)

Personally I'd probably prefer the second approach.

gavanderhoorn, 2018-06-25:
PR: #356.

gavanderhoorn, 2018-07-29:
Fixed by #356.

closed 2018-07-29 by gavanderhoorn